wled/WLED #4395: 'No option 'default_partitions' in section: 'esp32_idf_V4''

eku, 2024-12-14:
### What happened?

If using the file `platformio_override.sample.ini` as base for own `platformio_override.ini` compilation failes with

```
InvalidProjectConfError: Invalid '/home/erik/src/esp/wled/src/platformio.ini' (project configuration file): 'No option 'default_partitions' in section: 'esp32_idf_V4''
`````

### To Reproduce Bug

1. cp  platformio_override.sample.ini platformio_override.ini
2. build WLED_tasmota_1M

### Expected Behavior

No compile error.

### Install Method

Self-Compiled

### What version of WLED?

0.15.0 (HEAD)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
InvalidProjectConfError: Invalid '/home/erik/src/esp/wled/src/platformio.ini' (project configuration file): 'No option 'default_partitions' in section: 'esp32_idf_V4''
```

### Anything else?

The section `[env:esp32dev_V4_dio80]` from `platformio_override.sample.ini` references `esp32_idf_V4.default_partitions` which is nowhere defined.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-12-15:
hi @eku,

thanks for reporting the problem. I have looked at the example file, and fixed a number of problems where the example file was not aligned any more with the latest WLED 0.15 release.

https://github.com/Aircoookie/WLED/pull/4398

softhack007, 2024-12-16:
oops, gitbub closed the report automatically when merging.

eku, 2024-12-17:
> oops, gitbub closed the report automatically when merging.

It's because of your commit message `solves #4395`.

eku, 2024-12-21:
Thanks for fixing.

netmindz, 2024-12-22:
Is this resolved or not? Has the fix been merged to 0_15_0 too @softhack007 ?

pgy866, 2025-07-06:
[image]

I cloned the latest source code, the same error occurred, the merger fixed it?
2025.07.06   22:51

eku, 2025-08-09:
At e3653baf7443dec65119798bdb69d44fefb5cd20 I get again the error:

```
InvalidProjectConfError: Invalid 'wled/src/platformio.ini' (project configuration file): 'No option 'platform_packages' in section: 'esp32s2''
```

closed 2025-08-09 by eku