wled/WLED #4396: Weird flicker when toggling on/off Athom bulbs

Andro-Marian, 2024-12-14:
### What happened?

When the fade-in starts and when fade-out is finishing. A weird flicked happens.
https://drive.google.com/file/d/1YgvZzD1sl7MFx1y-zQKS_JEvJhUdDfPS/view?usp=sharing

### To Reproduce Bug

Flash the  0.15.0-b6 or the release 0.15 to any Athom 7w or 15w.
Turn it on/off by the web interface.

Noticed a flicker when going on or off.

### Expected Behavior

To be smooth as before.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15-(b6/release)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-14:
I noticed this too when working on the new PWM outputs. It is related to fading value being offset from the starting and target value but I could not pin-point where this actually happens.

Andro-Marian, 2024-12-14:
Not sure what was changed on Beta 6. Thinking if the https://github.com/Aircoookie/WLED/issues/3196 was introduced first on b6

JorinL, 2024-12-16:
Hey there,
before opening a complete new - according to the video - it could be also my issue.

I think i might have the same thing - it "looks" pretty the same.

I have this problem on my 3 PWM strips.
When I cut the transitiontime it is gone - with transition it flashes when turning on before dimming up and when turning off while dimming down a flash and dimming further down.

I added a small video of the current situation.
https://github.com/user-attachments/assets/f6009672-1853-451c-b2fd-f404a0fe5acc

marcolino7, 2024-12-19:
Hi,
same behavoiur here, WLED 0.15 on PWM RGBW strip. I reverted back to 0.14 and issue gone

DedeHai, 2024-12-23:
I found one issue in the code related to CIE brightness curve calculation. Is the issue gone if you enable dithering in the PWM LED settings?

closed 2024-12-23 by DedeHai

JorinL, 2025-01-22:
Thanks for fixing :-)
Tried the Alpha 16. Works :-)

SuulsaK, 2025-02-06:
I don't know if I'm doing something wrong, but I've flashed several of my Athom bulbs with supposedly fixed alpha firmware and now the flickef is gone, but is replaced with nice slow transition at start, then sudden jump to full brightness. I've checked if i have correct pins set, the issue still persist.

commit 88738327 (2024-12-23): Merge pull request #4419 from DedeHai/PWM_brightness_fix

BUGFIX PWM brightness discontinuity (fixes #4396)