wled/WLED #4399: Apply at boot - change preset via json/HTTP API

FlohGro-dev, 2024-12-15:
I would like to change the preset that should be applied at boot via the json/HTTP API.

Background: I use WLED for a lamp with interchangeable covers (e.g. a lion and a car) - I have different presets for the covers and I don't want to manually reconfigure the applied preset at boot every time I change the cover.

dosipod, 2024-12-16:
@FlohGro-dev You can use the end point http://wled-ip/json/cfg
[image]

FlohGro-dev, 2024-12-16:
Interesting! I couldn't find this in the docs either it's not there or I used the wrong search terms. Will try and close the issue if it's working 😊

dosipod, 2024-12-16:
@FlohGro-dev Yes it is not in the docs because no one bothered to updated it , once you tested it ( it will work  ) then you might want to consider making a PR for the docs here https://github.com/Aircoookie/WLED-Docs .

The http://wled-ip/json/cfg can work with all configuration based  items and if you find any that does not work then report it

FlohGro-dev, 2025-01-12:
@dosipod sorry for the late response. I didn't look too deep into it to understand where to put the documentation as a PR (which file).
Since the WLED project was a gift I can't work on that anymore for now. Thanks for your support here and hope someone else can add it to the docs sooner or later 😊

closed 2025-05-20 by github-actions[bot]