wled/WLED #4400: Request for PWM_fan or to implement safety temp threshold which turns LEDs off

orbitinstasis, 2024-12-16:
Similar to the (recently patched) internal temp mod, it'd be useful if the pwm_fan mod (or 'Temperature' mod) has the optional feature to turn the LEDs off if the temp hits a userdefined threshold. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Much like the internal temp mod, a 'preset to active' (preferably worded better) text box that takes in a preset to active once the 'activation threshold' user defined temp is hit would be useful since the probe I have setup is much closer to the enclosures probe, and therefore is far more useful and important to monitor.

softhack007, 2024-12-16:
>  preset 0 now correctly turns all LEDs off, and this would be the default desired behaviour.

@orbitinstasis just for my curiosity - why does turning LEDs off reduce the box temperature? The esp32 is not part of the LEDs power train (hopefully), so i would only expect a minimal effect on heat reduction from the esp32.

From my software perspective:
* not calculating effects - slightly reduces cpu load, but no thermal difference expected
* reducing Wi-Fi power output, or allowing wifi sleep, might help to reduce chip temperature
* Running at reduced CPU and Flash speed might help, however this requires a custom build (80MHZ CPU, 40MHZ flash)

orbitinstasis, 2024-12-16:
> > preset 0 now correctly turns all LEDs off, and this would be the default desired behaviour.
> 
> @orbitinstasis just for my curiosity - why does turning LEDs off reduce the box temperature? The esp32 is not part of the LEDs power train (hopefully), so i would only expect a minimal effect on heat reduction from the esp32.
> 
> From my software perspective:
> 
> * not calculating effects when LEDs off - slightly reduces cpu load, but no thermal difference expected
> * reducing Wi-Fi power output, or allowing wifi sleep, might help to reduce chip temperature
> * Running at reduced CPU and Flash speed might help, however this requires a custom build (80MHZ CPU, 40MHZ flash)
> 
> Actually there are other ways to reduce CPU load and (maybe) reduce temperature:
> 
> * Setting a lower "Target FPS" in led settings will give the processor more time to do nothing (i.e. cool down)
> * Disabling usermods that cause background activity (like audioreactive, four-line-display, rotary, ...)
> * Disabling standard features that cause background activity: serial protocols (WLED_DISABLE_ADALIGHT) and websockets (WLED_DISABLE_WEBSOCKETS), maybe even more
> * Not using HomeAssistant integration

Thanks for the tips with reducing power on the esp32 - they're useful but i think it's not quite the angle I'm approaching things. 

Off the top of my head:

tl;dr I am not worrying about the actual electronics melting - I am worried about the LED fixture / LEDs overheating. I usually add temp sensors in my projects to make sure my ambient temp remains sane, and it'd be really nice to have a feature with your mod to turn the LEDs off before the ambient temp hits a temp where the LEDs are generally too hot for long term use, and more importantly, to stop any plastic fixtures from melting.

Longer: For small projects (under ~20W) it's common to have the electronics hidden somewhere near the lights, furthermore, it's not uncommon for the LEDs to be adhered to some type of plastic (never PLA, but maybe PETG/ASA or other plastics with higher glass temperatures).  

A) for projects where the LEDs adhere to plastic,  I want to keep the temp of the plastic below its glass temp. I know there are many ways to do this primarily with current limiting - but it seems to make a lot of sense to have a feature maybe at least on the 'temperature' mod to have a cut off threshold (optional maybe).   

Currently I have a project which is entirely plastic with a glass temp of 80c, inside a lampshade is a tube, the LEDs adhere to the outside, the electronics are on the inside, and there are appropriate intake vents with a fan on the top, currently the temps are fine, but if ever the fan stops working etc there's no safety function fallback to turn the LEDs off if the temp dramatically increases. fyi the dallas sensor is inside the tube and a great representative temp for the plastic fixture (verified with a thermal camera). The temp I am keeping below (~44c) is far below the electronics threshold temps, but if the tube was made out of a metal, I'd still think this feature would be beneficial as a safety fallback.

blazoncek, 2024-12-16:
As an author of PWM usermod I can say that it does not behave in an on/off fashion.
When a temperature (if configured and enabled) is set it will increase PWM duty the larger temperature difference is.
PWM duty can be set without temperature usermod from UI (or UI can override temperature).

I.e. if you set 30ºC as a set temperature, the PWM output will have `minPWMValuePct` (in %) duty set if the temperatue is 30ºC or below. For every 0.5ºC above it will increase PWM duty by `step` amount untill 100% duty is achieved.
I would assume it is not desirable to trigger preset on the 1st threshold (30ºC) let alone have only one preset.
Adding hysteresis would be needed but that will increase complexity beyond what I'm willing to do.
But the question remains how to handle user interaction from UI in such case.

orbitinstasis, 2024-12-16:
> As an author of PWM usermod I can say that it does not behave in an on/off fashion. When a temperature (if configured and enabled) is set it will increase PWM duty the larger temperature difference [...]

I understand what you're saying behind the functionality of the PWM mod, i have been using it successfully and am happy with it. 

What I'm requesting wouldn't change any functionality of the pwm stage, but rather would add a seperate threshold temperature which when it, would turn all LEDs off - practically identical to how the 'internal temp' mod works. It is why i now wonder if it makes more sense to request such a feature to be added to the 'temperature' mod and not the pwm mod.