wled/WLED #4401: Some LEDS stay on after turning off WLED with Dissolve effect

cryptk, 2024-12-16:
### What happened?

When using the dissolve effect on my esp8266 based Dig-Uno controller, turning off WLED leaves some of the leds running. I am unable to reproduce this on my other controllers, but they are all esp32 based, so either I have a bad esp8266, or a bad dig-uno, or this issue is something specific to the esp8266.

### To Reproduce Bug

1. Use an esp8266 based controller with WLED 0.15.0
2. Configure a WS281x LED output with 50 LEDs
3. Use the following preset JSON:
```json
{"on":true,"bri":32,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":50,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"","col":[[255,0,0],[8,255,0],[0,0,0]],"fx":18,"sx":128,"ix":255,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```
4. Run the controller with that preset
5. Press the power button in the top left of the UI
6. Observe that some of the LEDs stay on (not sure if this happens every time, but it definitely seems to happen more often than not

### Expected Behavior

All LEDs should be powered off when the controller is turned off

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

One thing that I have observed is that on my esp32 based controllers, when turning the controller off with the dissolve effect active, the whole strip fades down to being off together.  On my esp8266 based controller the behavior is different, it looks like the effects "dissolves to off" if that makes sense.

Rather than the whole strip fading down and turning off, the LEDs turn off one at a time randomly, the same way the dissolve effect would transition from one color to the next, they are just dissolve transitioning to off.  It seems like while this is working, the controller just doesn't wait for the "dissolve to off" to complete, so a few LEDs get left on when the controller is fully powered down.

On my strip of 50 LEDs, it's typical for 2-5 LEDs to be left on if it is turned off while the dissolve transition is running.

Ideally, I would think the behavior should be the same between the esp32 and the esp8266, it should do a fade to off rather than a dissolve to off.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

megachammer, 2024-12-19:
I have the same issue with both esp8266 and esp32, but I cannot provide a solid way of how to reproduce this. It just happens from time to time with all of my boards.

cryptk, 2024-12-23:
> I have solved this by enabling the "Off Refresh" option in LED settings, I'm using a Wemos D1 though (esp8266 based) However now the blue onboard led doesn't turn off anymore, but that's another issue

I'll give that a try tomorrow and do some testing

blazoncek, 2024-12-23:
Do you use ledmaps?

cryptk, 2024-12-23:
> Do you use ledmaps?

No, it's just a simple string of 50 RGB LEDs that I lay out for Christmas and Halloween. It's running a playlist with a few effects in it and I have only noticed the issue when I turn it off with the dissolve effect active. 

The only adjustments I have made to any of the effects are color, speed and intensity.

blazoncek, 2024-12-25:
Only happens on ESP8266 ant it is even more pronounced if you set transition time to 0.

blazoncek, 2024-12-25:
Found the issue: Not all pixels are updated during `strip.service()` call and hence some will not obey `strip.setBrightness()` call.

blazoncek, 2024-12-28:
This issue is present on all effects that do not update every pixel on each `strip.service()` call.
Some will behave better than others, though.

PR #4428 (2025-01-16): FX: Waterfall, Matripix & Dissolve fix
- for Arc expansion
- or gaps
- or ledmaps with missing pixels

Fixes #4416
Fixes #4401