wled/WLED #4404: 0.15.0 sync issues

jjvelar, 2024-12-17:
### What happened?

Since updated to 0.15.0, sync is not working anymore.

### To Reproduce Bug

I've got one WLED configured to SEND, and another one configured to RECEIVE.
Change color or preset on SEND WLED and nothing changes in RECEIVE WLED.

### Expected Behavior

Sync feature should work.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-17:
https://wled.discourse.group/t/sync-question/13615

megachammer, 2024-12-19:
I have 4 boards, a mix of 8266 and ESP32. One of them is the main board. Sometimes (quite often, actually), only 3 out of 4 sync, and I need to send the same command (like a new effect) again to get all of them synced.
0.15.0

jjvelar, 2024-12-19:
> https://wled.discourse.group/t/sync-question/13615

Thanks but this didn’t work for me

DedeHai, 2024-12-20:
From current information it is hard to say if this is something to do with configuration or an actual issue with the code. Can you provide more details like your exact settings that reproduce this issue?

netmindz, 2024-12-20:
If you are seeing it work intermittently, then I would recommend you increase UDP packet retransmissions to help overcome any issues with your WiFi

jjvelar, 2024-12-20:
It is not working at all in my case.
In any case, tried 30 and didn’t work.

ShellBoyPhotography, 2024-12-21:
I have exactly the same problem and tested it in three different WLANs:

Huawei router/WLAN and Ubiquiti/WLAN as well as AVM Fritz.Box/WLAN the controller is an ESP32 

I have reset the controllers (8 in total) to factory default and also set the UDP packet repetition rate to 16 or 32. No sync.

I still have two controllers with 14.4.4 that sync perfectly in all WLANs. 

I would say the problem is the new firmware and not the WLAN settings
It may be due to the ESP32 version if this problem does not occur for all users.

ShellBoyPhotography, 2024-12-21:
it's a bug.
i downgrade my controller and it works fine for me. 
your welcome :/

netmindz, 2024-12-22:
The key word in my message was _intermittently_ - i.e if changes sync, but not every time.

If no changes are working then you probably need to review the settings of what values are sent in sync as there are new options in 0.15

E.g

Send notifications on direct change
Send notifications on button press or IR

jjvelar, 2024-12-22:
Downgraded to 0.14.4 and sync is working again.

ShellBoyPhotography, 2024-12-22:
Oh i’am sorry, I think I have reading a thread from top to bottom. Anyway, I read that too and also undid the new changes in the 0.15.0 firmware.

And still the error remains. But I just changed it again to take my tomatoes off my eyes and changed the firmware of two controllers to 0.15.0.

Tried everything described "here" and in the link (https://wled.discourse.group/t/sync-question/13615) and it remains the same as before. 

Anyway. Xmas is in 2 days here in Germany and I like my christmas decoration in sync. Sorry 😢! A Downgrade is possible and works for me and any other problems will be fixed or ignored. Whatever. 

Never mind.

Maybe I'll have a look at the source code between the holidays, maybe it's the .bin file and I'll create my own. However: Advantage of OpenSource DIY and feel free

have fun, no hate and have a good vacation
and Merry Christmas 

Good New Year's resolutions: learn more fishing less trolling

netmindz, 2024-12-22:
Merry Christmas @ShellBoyPhotography 

If you could share screenshots of the sync settings for both the sender and receiver and also give step by step of your actions please. E.g are you making changes on the master with the web UI, buttons, HA automations, IR remote, ESP-NOW remote etc

ShellBoyPhotography, 2024-12-22:
Yes of cause @netmindz 
But after Xmas bro 👊

raunak51299, 2024-12-23:
Turing on "Send notifications on direct change: " in sync interface fixed my issue.

jjvelar, 2024-12-23:
> Turing on "Send notifications on direct change: " in sync interface fixed my issue.

This didn't work for me.

netmindz, 2024-12-23:
Please share screenshots of your sync config @jjvelar

blazoncek, 2024-12-23:
FYI sync configuration changed in 0.15 (noted in chnage log).
You need to enable sync events on the master device, i.e. "direct change", "button", "alexa", etc.

jjvelar, 2024-12-23:
all of them?

jjvelar, 2024-12-23:
> only the ones you want.

I had one activated and didn't work for me.

blazoncek, 2024-12-23:
Not the correct one then. 😉

jjvelar, 2024-12-23:
Might be :-)
Using “send notification on direct change”
Should I use “send macro notifications” or any other?

jjvelar, 2024-12-26:
Hi, sorry for the late reply.
here you've got screenshots of my configuration:

Receiver:
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

Sender:
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

netmindz, 2024-12-26:
None of those images have come through @jjvelar, you need to use the GitHub website rather than replying to the email

jjvelar, 2024-12-27:
Hi, here you've got the screenshoots again:

Receiver:
[image]
[image]
[image]
[image]

Sender:
[image]
[image]
[image]
[image]

jjvelar, 2024-12-28:
Any thoughts based on the screenshots? Thanks!

closed 2025-05-05 by github-actions[bot]

netmindz, 2025-05-10:
Nothing immediately jumps out at me from that, but then I don't really use sync that much myself so I might be overlooking something 

What's your network setup, can both see each other in the nodes list?