wled/WLED #4405: Virtual LED LAG and userinterface access issues

jonny190, 2024-12-18:
### What happened?

2 or 3 times a day i need to restart a dedicated WLED isntance that is used to controll 5 other WLED instances with the actual LEDS connected to them totalling 1072.

The LEDS are almost always set to rainbow pattern and the pattern starts to be unstable on the output instances and equally i loose access to the webUI

[image]

### To Reproduce Bug

Rainbow Color
Rainbow Pattern
using all 1072 LEDS as a giant strip

### Expected Behavior

Pattern to flow smoothly 

### Install Method

Binary from WLED.me

### What version of WLED?

0.15-b7 2410270

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
Post device reset using insatall.wled.me the output is 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f6996
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (272) esp_core_dump_flash: No core dump partition found!
E (272) esp_core_dump_flash: No core dump partition found!
E (406) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
Ad
```

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-18:
`USB_UART_CHIP_RESET` may be a hint. what is connected to the USB?

jonny190, 2024-12-18:
@DedeHai is that not because I triggered a device reset via the install.wled.me page to get some form of logs, as apart from that I get no log output

DedeHai, 2024-12-18:
you need to install a debug version to get more output.
could be several things:
- unstable power supply
- home assistant
- memory leak (check available heap from time to time to make sure it does not get smaller)

jonny190, 2024-12-18:
I've tried powering via a Meanwell 5v PSU, Via a 18650 battery and via USB (both computer and phone charger)

Same issue occurs when Home Assistant has the device disabled

Ive just recompiled with the debug flags and while the symptoms are active took a dump of the logs while the webUI wasnt loading:

```
---DEBUG INFO---

[13:16:37]Runtime: 300020
Unix time: 1734527736,316
Free heap: 214332
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 2020
Loop time[ms]: 0/12
UM time[ms]: 0/1
Strip time[ms]:0/12
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
WS FileRead: /index.htm
[13:16:44]WS FileRead: /index.htm
[13:17:01]Local time: 13:16
Trigger time: 08:26
Trigger time: 15:40

---DEBUG INFO---
[13:17:31]Runtime: 330021
Unix time: 1734527766,317
Free heap: 218780
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1964
Loop time[ms]: 0/13
UM time[ms]: 0/1
Strip time[ms]:0/13
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B

---DEBUG INFO---
[13:18:01]Runtime: 360023
Unix time: 1734527796,319
Free heap: 205144
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1978
Loop time[ms]: 0/12
UM time[ms]: 0/1
Strip time[ms]:0/12
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Local time: 13:17
Trigger time: 08:26
Trigger time: 15:40

---DEBUG INFO---
[13:18:31]Runtime: 390024
Unix time: 1734527826,320
Free heap: 229096
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1972
Loop time[ms]: 0/12
UM time[ms]: 0/1
Strip time[ms]:0/12
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B

---DEBUG INFO---
[13:19:01]Runtime: 420027
Unix time: 1734527856,323
Free heap: 227676
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1955
Loop time[ms]: 0/12
UM time[ms]: 0/1
Strip time[ms]:0/12
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Local time: 13:18
Trigger time: 08:26
Trigger time: 15:40

---DEBUG INFO---
[13:19:24]Runtime: 450028
Unix time: 1734527886,324
Free heap: 227676
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1994
Loop time[ms]: 0/12
UM time[ms]: 0/1
Strip time[ms]:0/12
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
WS FileRead: /index.htm
[13:19:31]---DEBUG INFO---
[13:19:32]Runtime: 480031
Unix time: 1734527916,327
Free heap: 208464
PSRAM: 1964kB/2045kB
TX power: 34/34
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 4686
Client IP: *.*.15.198
Loops/sec: 1963
Loop time[ms]: 0/19
UM time[ms]: 0/1
Strip time[ms]:0/13
Segments: 1 -> 68/0B
  Seg: 1072,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
WS FileRead: /index.htm
[13:19:32]WS FileRead: /index.htm

```

I'm not sure what to make of the logs and i cannot get to the webUI to get the available heap. 

Hopefully the above is help to someone as i'm at a loss

closed 2025-04-25 by github-actions[bot]