wled/WLED #4409: Virtual LED Universe

lnorton89, 2024-12-20:
**Is your feature request related to a problem? Please describe.**
Virtual ArtNet RGB(W) LED's have no option for universe configuration.

**Describe the solution you'd like**
While adding virtual LED's for ArtNet it would be nice to be able to select what ArtNet universe that virtual output is outputting to. Also adding a toggle for multicast would be great!

netmindz, 2024-12-20:
@troyhacks are either of these covered by your recent improvements?

lnorton89, 2024-12-25:
I'm looking over udp.cpp and wondering if this is where the universe could/should be added? I will await further input from someone who knows the codebase better :)

```cpp
...
case 2: //ArtNet
{
...
uint16_t baseUniverse = 0; // Set your desired starting universe here (e.g., 0, 1, 10, etc.)
...

// Set the universe by combining the baseUniverse and currentPacket
uint16_t universe = baseUniverse + currentPacket;
ddpUdp.write(universe & 0xFF); // Universe LSB
ddpUdp.write((universe >> 8) & 0xFF); // Universe MSB
...
```

blazoncek, 2024-12-25:
That would require additional changes in settings and bus management.

smatthew, 2025-04-15:
So, what do we need to do to accomplish this? I'm willing to help.

blazoncek, 2025-04-15:
> So, what do we need to do to accomplish this? I'm willing to help.

Write the code and test it.

lnorton89, 2025-04-16:
> So, what do we need to do to accomplish this? I'm willing to help.

I lost interest when trying to implement the UI controls. The way the DOM (HTML page) is tied to the rest of the codebase was really just too complicated for me.

troyhacks, 2025-04-16:
@lnorton89 WLED-MM has an extensive Art-Net overhaul that might be useful. 

There's no "multicast" Art-Net - it uses broadcast. To enable that, just set the IP to your broadcast address, usually x.x.x.255

My new Art-Net overhaul didn't have "starting universe" but I agree that should be added - so I just committed some code for that along with some other minor improvements to the "hints" text block.

I re-used the incoming Art-Net starting universe setting we already have for incoming data, as you're not likely to be using both at the same time... I hope, anyway.  I'll look at having separate universe starts for incoming and outgoing at some point, as it might be handy for controlling a WLED-MM instance via Art-Net that's also outputting Art-Net.

[image]

At the moment there's no easy way to have different LED counts on different outputs - ledmaps might help there tho.

Because Art-Net in big setups is broadcast to several listening devices, I wouldn't recommend having more than one Art-Net sender to unicast IPs defined, even if it technically works. Broadcast once to all of them and have different starting universes set on each device.

troyhacks, 2025-04-16:
> > So, what do we need to do to accomplish this? I'm willing to help.
> 
> I lost interest when trying to implement the UI controls. The way the DOM (HTML page) is tied to the rest of the codebase was really just too complicated for me.

10000000% agree. It took a lot of time to make this overhaul work with the GUI.

roynasser, 2025-10-19:
Hi guys, I have a setup I'd like to transition to WLED that is basically 4 corridors with DMX rgbww downlights in a straight line... (each "pixel" is 5ch rgbww) I'm thinking WLed would be a good system to control it after having used it at home.

Am I understanding correctly - with this output mode (ArtNet (network)) I can use any artnet compatible ethernet dmx output device to output the DMX to the line, and use WLed on an ESP32 connected to no leds. Is this correct?  Assuming it is - is there any way to run WLed on VM or Docker container? (I'm assuming not, as it requires an ESP32 chip)... but I would have prefered an emulated version so I could run a VM and not run the risk of the esp32 board crapping you...

thanks for the info!

troyhacks, 2025-10-26:
@roynasser this works, kinda.

I certainly run RGBW DMX lights this way - but 5-channel RGBWW would require a new mode and perhaps a new WW calculation method. This is not impossible.

I use an Art-Net to DMX512 devices all the time - I have ones with 2 and 4 DMX outputs, corresponding to 2 or 4 universes. (And beyond, some can do 8 or 12 or 16 DMX outputs, but mine don't have the nice XLR jacks on them. I think some are available that do tho.)

I wouldn't worry much about the ESP32 crapping out - most of my projects have run for weeks without issues. Stability is very good.

Mostly it's the RGBWW that's the blocker as currently it only works for RGB and RGBW (3 and 4 channels respectively).

roynasser, 2025-10-26:
Hi @troyhacks , thanks for clearing that up for me.  I'll give it a try soon.  I suppose you are right about the ESP reliability, esp without the power component running on the same board or box reliability should not be an issue.  And even then, I can have a cold-spare in place to be swapped in.

> Mostly it's the RGBWW that's the blocker as currently it only works for RGB and RGBW (3 and 4 channels respectively).

I suppose I could use it with only 4ch... Not the best case scenario but it definitely would beat the current control situation :)

Am I looking correctly that adding CCT would be around here? https://github.com/wled/WLED/blob/b60313e1f884e129197a7e08d8c669593fae01d3/wled00/bus_manager.cpp#L713