wled/WLED #4412: Random reboots when connected to wifi but not in AP mode

TheNutHouse84, 2024-12-21:
### What happened?

ESP32 based devices rebooting randomly after 5-10 minutes. Happens only when connected to local wifi but not in AP mode.  No leds connected. Happens on two different Gledopto GL-C-015WL, versions 0.14.4 & 0.15.0 and third bare NodeMCU-ESP32 flashed with v0.15.0. 

### To Reproduce Bug

Connect ESP32 based wled controller to Calix router. 

### Expected Behavior

Stable operation when connected to wifi with no random reboots

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0 & 0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Ada
abort() was called at PC 0x4014ef4e on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ad5c:0x3ffb1e40 0x4008b0b9:0x3ffb1e60 0x4014ef4e:0x3ffb1e80 0x401ba990:0x3ffb1ea0 0x401ba955:0x3ffb1ec0 0x4012bbf7:0x3ffb1ee0 0x4013812e:0x3ffb1f20 0x4011be4d:0x3ffb1f40 0x4011c5fa:0x3ffb1f90 0x4013d3a1:0x3ffb1fb0 0x4008cb42:0x3ffb1fd0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
Ada
```

### Anything else?

Home wifi is Calix Gigasphere GS-4220 router.  Same NodeMCU-ESP32 flashed with Tasmota 14.4 connected to same wifi router is stable over 2+ days without rebooting. Tasmota uses  ESP32/Arduino library Core version v3.1.0.241206

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TheNutHouse84, 2024-12-25:
> Please use debug build with exception decoder. Otherwise that crash report does not tell much.
> 
Is there a downloadable .bin or do I have to compile?

> Tasmota uses very different environment than WLED. Don't compare apples to pears.

Not trying to compare, just trying to provide as much info as possible.

blazoncek, 2024-12-25:
You will need to compile yourself if you want to use exception decoder.

Impedanz, 2024-12-25:
I have a similar problem, WLED starts, resets or crashes in the middle of operation or at night (switched off) and then switches on automatically although the check mark is not set (Enable after Reset /Power Loss).
Error also occurred previously in the beta. I have already tried ESP32 4, which allows me to rule out a hardware defect. If it was in operation, the light is simply reset, otherwise the light simply switches on at night. I would be happy if the error is found, because otherwise it switches on when nobody is at home.

TheNutHouse84, 2024-12-25:
I flashed 0.15.0 to an ESP32-C3 I have (as a comparison test) and that has remained connected to my wifi for over 4 hours now.  I have yet to compile a debug version for the standard ESP32-WROOM.

Impedanz, 2025-01-19:
Updated to 0.15.0 on ESP32, Error still happen (WLED Crashes and Reboots)

DedeHai, 2025-01-20:
@Impedanz are you able to provide a crashdump with exception decoder? without a crashdump there is nothing that can be done.

DedeHai, 2025-01-20:
> I'm sorry, how do I create a crash dump of this?

I don't know if there is any good tutorials, found this:
https://randomnerdtutorials.com/vs-code-platformio-ide-esp32-esp8266-arduino/

but instead of upload, use upload and monitor:
[image]

and at the monitor output, you will get the crash dump if it crashes.

I also found this: https://quinled.info/wled-crashes-and-reboots-every-xx-minutes-mdns/