wled/WLED #4417: Back button on Update.htm

Jacobnordvall, 2024-12-22:
### What happened?

Back button dosnt go back to index if the last page wast index. aka you clicked on git banner to get latest bin.

### To Reproduce Bug

Click the git banner. then use back gesture on android. then use the back botton. It now takes you back to the git page as it tracks the last page and dosnt actually go to index page.

### Expected Behavior

back goes back to last entry in the main page.

### Install Method

Self-Compiled

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

Here is a fully working solution for reference:
`function goBackToMainPage() {
    let stepsBack = 0;

    function checkHistory() {
        const currentPage = window.location.href;

        // Define conditions for your "main" pages
        if (currentPage.includes('/#Colors') || 
            currentPage.includes('/#Effects') || 
            currentPage.includes('/#Segments') || 
            currentPage.includes('/#Presets')) {
            return true;
        }
        return false;
    }

    function navigateBack() {
        if (stepsBack < 20) { // Prevent infinite loop
            stepsBack++;
            window.history.back();
            setTimeout(() => {
                if (!checkHistory()) {
                    navigateBack();
                }
            }, 100); // Wait for navigation to take effect
        } else {
            // Fallback if no "main" page is found
            window.location = '/';
        }
    }

    if (!checkHistory()) {
        navigateBack();
    } else {
        // If already on a main page
        console.log("You're already on a main page!");
    }
}`

and ofc the html:
`<button type="button" onclick="goBackToMainPage()">Back</button>`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jacobnordvall, 2024-12-22:
that code (or a similar fix) should also be used for the settings back button preferably. as it goes back to the actuall last index page and not just to colors no matter what.

blazoncek, 2024-12-23:
Make a PR, please.

Jacobnordvall, 2024-12-24:
> Make a PR, please.

Want me to put that logic in common.js? 
As it can be used for proper back functionality for settings page too and others?

DedeHai, 2024-12-24:
keep in mind that flash memory is at a premium, keep the code as compact as possible even if it means cutting corners, IMHO this is solving an inconvenience which currently minimal code should be spent on.

blazoncek, 2024-12-24:
Also verify functionality behind reverse proxy.

Jacobnordvall, 2024-12-26:
> Also verify functionality behind reverse proxy.
im not entirely sure what that means tbh.
its a line swap fix. if its just to adress the bug.

[Pull request](https://github.com/Aircoookie/WLED/pull/4431)

blazoncek, 2024-12-27:
The behaviour is correct and expected one. When you click/tap Github link, a new tab/window is opened. It has no history and so cannot go back while the Update page correctly returns to previous window.

Jacobnordvall, 2024-12-27:
> The behaviour is correct and expected one. When you click/tap Github link, a new tab/window is opened. It has no history and so cannot go back while the Update page correctly returns to previous window.

Its not. Its broken in the wled app.

blazoncek, 2024-12-28:
Then open an issue in WLED app.
I am truly sorry for you but WLED behaves correctly and as designed.

Jacobnordvall, 2024-12-29:
Well this isnt going to be fixed with an app update.
Its wled not working well. And besides. Its broken in ap mode too. And the current way of handling is less performant too. I can create a pr to fix it all universally in wled but im guessing its no point as you see this behavior as intended?

closed 2024-12-29 by Jacobnordvall