wled/WLED #4420: JSON api: Wrong Preset and Playlist values in http verbose response

TheNetStriker, 2024-12-23:
### What happened?

I'm using OpenHab to control my wled controllers. When activating a preset or playlist OpenHab sends the "v": true parameter to get the selected values back in the response. But in the response the "ps" value is always the previous value and the "pl" value is always -1. This causes the values in OpenHab to always jump back to the values the wled controller returns.

### To Reproduce Bug

Send e.g {"ps": "1", "v":true} to the /json/state endpoint.

If ps value is a preset the returned "ps" value is the previously selected value.
If ps value is a playlist the returned "ps" value is the previously selected value and the "pl" value is always -1.

Sending the same request again results in the correct "ps" value, but "pl" is always -1.

For some reason the "pl" value in the response is updated when first sending the playlist id with the "ps" attribute (e.g. {"ps": "3", "v":true}) and then sending the same id via "pl" attribute. (e.g. {"pl": "3", "v":true}) But I guess this should not happen because the manual says that the "pl" attribute is read only.

### Expected Behavior

Values in the verbose response should return the "ps" and "pl" value that was set in the request.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-25:
Presets (and hence playlists) are loaded asynchronously which means using "v" to return preset will inevitably return currently selected preset not the new/intended one.
Upon actually loading it the response value will return correct preset ID.

You need to subscribe to websockets to get an update when preset actually changes.

TheNetStriker, 2024-12-25:
@blazoncek Thanks for the hint. I will try to find for a solution in the OpenHab binding for this.

But what is the situation with the playlist value? Why is it always -1?

blazoncek, 2024-12-25:
Playlist is -1 when it is unloaded, i.e. no longer active.

As for JSON API, there is no "pl" key, only "ps". "pl" is only available in response as a read-only property. See [KB](https://kno.wled.ge/interfaces/json-api/).

TheNetStriker, 2024-12-26:
@blazoncek Ok thanks, I see now how it works. As long as I'm sending the "ps" value to set the playlist, the value returns as -1. But the next time when I'm only querying the values the "pl" value is returned correctly.

I will try to find a solution for this in the OpenHab binding. I'm closing this issue now.

closed 2024-12-26 by TheNetStriker