wled/WLED #4433: Support for all in one ESP32-S3 Matrix 8x8 devices using GPIO14

joshcliffejones, 2024-12-26:
This is a feature request for the support of WLED on commercially available all in one ESP32 LED matrices.

I recently purchased a couple of these items https://www.aliexpress.com/item/1005006962940633.html?spm=a2g0o.order_list.order_list_main.16.44d31802hXHO91

And had simply planned to use them with ESPHome as they can be used as an array of lighting or text scrolling options.

However, I thought I would try running WLED on them, but though I got WLED installed and running, it wouldnt then let me make use of GPIO14 which these boards use as their output.

Thank you for your continued work on WLED, it is such a fantastic project that I use extensively! 👍 

You should create a Patreon for anyone wishing to make a regular contribution for your efforts on the project!

DedeHai, 2024-12-26:
there is no restriction on GPIO14, check your settings (usermod settings). If you check discord, you can find people using this ESP already.
For further support questions please use discord or the discourse forum.

closed 2024-12-26 by DedeHai

softhack007, 2024-12-26:
@joshcliffejones I'm not sure why you think these boards are not supported? If you cannot assign GPIO14 to a led output, it usually means the pin is already configured for another function, like IR, Relay or maybe for audioreactive. 
You just need to go through the settings pages (usermods, too), find where GPIO14 is used, and set it to "-1" or "unused".

Please use discord to get further help from our users community 

[image]

Making an enhancement request that's actually just a question "how to make it work on my board" will not help you. But our discord community will be happy to help.