wled/WLED #4437: Color Order Override does not work with DDP

dosipod, 2024-12-28:
### What happened?

We have highlighted this issue during the last fix for Color Order Override bug here 
https://github.com/Aircoookie/WLED/pull/4326  but that was ignored . 

Color Order Override works with DDP in MM so it is there for the taken 

### To Reproduce Bug

Setup DDP and use Color Order Override   with both  MM and AC  , below pic is from MM  . It does not work in AC
[image]

### Expected Behavior

Color Order Override to work with DDP 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.16.0-dev (build 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
NA
```

### Anything else?

Fixed in MM by @troyhacks  along with other improvements to artnet output .

It would be ideal that anything done in the future  for local leds to consider DDP 
as  that would be the expected behavior since it is an output 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-30:
IMO virtual LEDs should have color order override and should always be considered RGB(W). Why? Because there is no hardware attached to ESP which is the one that can have different color order.

The color order makes sense only on physical LEDs, and if you configure receiving device (the one with LEDs attached) correctly no additional configuration is necesary on the sending device.

This is even more important if you replace a receiving device with different color order than original one. If color order was defined on the sender, you would need to make adjustments there.
And all external SW would need to take that into account, i.e. xLights.
Also local effects would play incorrect colors on such device.

All that said, yes technically it is possible to have COO even for virtual LEDs but it has been left out intentionally for the reasons stated above.

dosipod, 2024-12-30:
@blazoncek  Although you may have noticed we post a lot on DDP , the real aim for us with DDP  is mainly for beta testing  as we can not keep re-flashing or testing  on other permanent units having physical leds so DDP  not behaving  the same as physical leds output  would simply make testing harder or prevent it so I hope you get the idea . Seeing that in MM was the reason we highlight it  again

netmindz, 2025-01-08:
You mention that is doesn't work in AC. but I don't see it in AC as an option and I agree with @blazoncek that it only makes sense for output 

Sender should send RGB then if receiver has a mix of different LEDs you map to the correct order

dosipod, 2025-01-08:
@netmindz The receiving devices  we have are not touched ,some with old builds ,custom builds ,( even some still with the color order bug we know about )  and we do not wish to change their firmware or setting hence we use DDP but it is not really only that at this point  as seeing that in MM must mean there are use cases for it .From our side we prefer to have everything that is using DDP behaves exactly the same as real output which is not limited to color order  but as said it is not critical

closed 2025-01-08 by netmindz

dosipod, 2025-01-08:
@netmindz I hope you are not targeting my posts  , please do not close any of them , you can ask me to close them as by doing so you are saying that the way it is done in MM is the wrong way

netmindz, 2025-01-08:
No offence meant. I clearly misinterpreted your comment about not being critical

dosipod, 2025-01-08:
Thank you , I have kept some issues open as we are working to see if we can fix them from our side if no one else has the time to do so  , I am having hard time testing other guys work  and also doing PRs from our side  so I hope you understand