wled/WLED #4438: ESP32 update from 14.4 to 15 not possible

PLCHome, 2024-12-28:
### What happened?

When installing the WLED_0.15.0_ESP32.bin via the web interface, the message "Update failed!
Please check your file and retry!" appears.

### To Reproduce Bug

Install WLED_0.14.4_ESP32.bin on an ESP32 then upgrade to WLED_0.15.0_ESP32.bin via the web interface.

### Expected Behavior

Install WLED_0.14.4_ESP32.bin downgrade via web interface on an ESP32 then upgrade to WLED_0.14.3_ESP32.bin and WLED_0.14.2_ESP32.bin is no problem from WLED_0.14.2_ESP32.bin to WLED_0.14.4_ESP32.bin is no problem. Update WLED_0.15.0_ESP32.bin via the web interface does not work.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.15.0_ESP32.bin

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-28:
most likely an incompatible file system. how did you install WLED originall? official web installer or using tasmota or home-assistant?

PLCHome, 2024-12-28:
I downloaded all the files here  https://github.com/Aircoookie/WLED/releases/tag/v0.15.0. Everything works up to 14.4, everything up to 15.0 doesn't work, including beta and rc.

softhack007, 2024-12-29:
@PLCHome how did you install your first WLED (initially = before OTA update? 

Did you use https://install.wled.me, https://wled-install.github.io/, or a different installation method?

closed 2024-12-29 by PLCHome