wled/WLED #4440: JSON Remote Preset Control Stops at ID 5

Sanjay7549, 2024-12-29:
### What happened?

# WLED JSON Remote Issue: Preset Changes Stop After ID 5

I am using the latest version of WLED (v0.15.0) on an ESP32, with a TSOP (IR) sensor added. I encountered an issue while using the JSON remote functionality. 

## Issue Description:
In my JSON configuration, I have the following commands for increasing and decreasing the preset:

```json
"0xF730CF": {
  "label": "Presets",
  "cmnt": "Increase by 1",
  "cmd": "PL=~1"
},
"0xF7B04F": {
  "label": "Presets",
  "cmnt": "Decrease by 1",
  "cmd": "PL=~-1"
}
```

However, when I press the corresponding buttons on the remote, the presets change only up to ID 5. After that, the preset stops changing, even if I press the button multiple times. The effect remains stuck at ID 5, and no further changes occur despite having more presets available.

### To Reproduce Bug

The following JSON configuration defines two buttons with IR codes for increasing and decreasing a preset. There is a bug that needs to be reproduced based on these configurations.

### JSON Configuration for IR Codes

```json
{
  "0xF730CF": {
    "label": "Presets",
    "cmnt": "Increase by 1",
    "cmd": "PL=~1"
  },
  "0xF7B04F": {
    "label": "Presets",
    "cmnt": "Decrease by 1",
    "cmd": "PL=~-1"
  }
}

### Expected Behavior

The system should cycle through all saved presets ( i have 17 saved presets) seamlessly, reaching the last preset ID without any issues.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-12-29:
Yeah we seen this limit of 5 if you use PL=~1, the way  it worked for example if you saved 17 presets is  use: 
PL=~P1=1&P2=17

blazoncek, 2024-12-30:
The behaviour is by design. Use a workaround provided by @dosipod .

closed 2024-12-30 by blazoncek

RestNPizza, 2025-01-21:
> The behaviour is by design. Use a workaround provided by [@dosipod](https://github.com/dosipod) .

Could you explain why this behavior is designed this way? The way the WLED docs read, the PL=~ should cycle through all presets, not just 1 through 5 so this is either a bug or the docs need to be updated (See the "In-/decremeting values" section here: https://kno.wled.ge/interfaces/http-api/)

In my opinion, it would be much more beneficial if the PL=~ command cycled through all preset numbers and then wrap around to the first once the last one is reached just like it says it should in the docs.

The problem with the work around that @dosipod listed is you have to change the ir.json file if you ever add a new preset. I have a scenario where i've made an LED sign for a family member that includes an IR receiver and remote as I dont expect them to use the app a whole lot. I have set up some presets already and would like to have a remote button cycle through all the presets listed. If I have to define which presets it cycles through, if my family member ever does use the app and adds a new preset, the remote will not include it in the cycle. They unfortunately are not technologically savvy enough to change the ir.json code on their own. I know I can make the preset range wider to begin with but the issue there is if it gets to the last listed preset and there are still 20 left in the predefined range, it wont cycle back to the first until the button is pressed 20 more times which is just a poor user experience not to mention confusing since it looks like the button stops working.

Would it be possible to re-open this issue and update the functionality to match that listed in the HTTP Request API?

blazoncek, 2025-01-21:
If you do not include P1 and P2 that's the *default* behaviour. Want more? Include P1 & P2.
`ir.json` is easy to change and upload. Examples available on KB.

If you think KB needs updating, please do so by submitting a PR to [WLED-Docs](https://github.com/Aircoookie/WLED-Docs).

RestNPizza, 2025-01-21:
> If you do not include P1 and P2 that's the _default_ behaviour. Want more? Include P1 & P2. `ir.json` is easy to change and upload. Examples available on KB.
> 
> If you think KB needs updating, please do so by submitting a PR to [WLED-Docs](https://github.com/Aircoookie/WLED-Docs).

Thank you for your reply and for all the incredible work you and everyone who works on WLED put into this project.

I wanted to ask if there’s a specific reason or limitation that causes this to be the default behavior? I understand that modifying ir.json is straightforward for someone who knows what they’re doing. But it’s a bit like flying a plane...it’s easy if you know how to do it, but most people don’t. For users like me who want to incorporate WLED into projects where the end user isn’t likely to know how to modify configuration files, this default behavior can feel limiting.

From my perspective, changing the default behavior to cycle through all presets when a range isn't defined would be more intuitive. It would also align better with other API commands that allow incrementing or decrementing values. As far as I know, none of those have default behaviors that are similarly restrictive, like cycling through only the first five values. This consistency could make things easier for a broader range of users.

I’ve also seen this point raised by others in discussions on Discord, forums, and Reddit while researching this issue in the past few months. It seems to be a common area of confusion, which suggests an opportunity to smooth out this aspect of the user experience.

Once again, I want to stress how much I appreciate the work everyone on the WLED team does. My goal here isn’t to criticize but to offer some constructive feedback on an area that might help polish the experience for users. Thank you for taking the time to consider this suggestion! If this is still not something you'd be willing to change, could you point out to me where in the code this default behavior is defined?

Thanks again!

blazoncek, 2025-01-21:
The limits are there since forever. 😄 They also correspond to 5 macros original versions of WLED had (I think, that was before my time).
WLED now features presets which do not exist by default and hence no values could be default.

Also, HTTP API is not evolving, JSON API is.

One more thing, if you have gap in preset, HTTP API cycling may not work.

RestNPizza, 2025-01-21:
> The limits are there since forever. 😄 They also correspond to 5 macros original versions of WLED had (I think, that was before my time). WLED now features presets which do not exist by default and [...]

Wow, you're fast!

You're right...I did see that HTTP API is no longer being worked on. Is there any chance a new command to cycle through all presets could be incorporated in the JSON API? (without the need to define a preset range of course) 😁

blazoncek, 2025-01-22:
No. Unfortunately.
Due to WLED not *knowing* how many presets there are. UI does, but firmware does not.

RestNPizza, 2025-01-22:
> No. Unfortunately. Due to WLED not _knowing_ how many presets there are. UI does, but firmware does not.

Gotcha. Thank you very much for your explanation and help!