wled/WLED #4441: WT32-ETH01 GPIO14 no longer available

joelhauxwell, 2024-12-29:
### What happened?

I upgraded to 0.15.0 and now there is an issue with me using GPIO14 for an LED string-

[image]

[image]

I noticed this as after the update, the brightness limiter was on by default - i'm not able to turn the brightness limiter off as the page complains about the output when i click save.

The output still works, its all just very dim due to the imposed 650mA limit.

### To Reproduce Bug

This has affected all my WT32-ETH01s which have been upgraded to 0.15.0

### Expected Behavior

I expect GPIO14 to still be available (as i have custom PCBs installed in my son's bedroom!)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FreaknSweet, 2024-12-29:
GPIO14 is default assigned in the user mod menu, unassign it from the user mod that is using it and it will become available for output.

joelhauxwell, 2024-12-29:
Thankyou so much - the user mod menu was a new one on me!

closed 2024-12-29 by joelhauxwell