wled/WLED #4443: TinsyS3 wont start on version 15 or later

jesse-schein, 2024-12-30:
### What happened?

I cant get my tinys3 board to boot on the latest versions of wled after making a new board config. It wont connect to wifi or show an access point after upgrading or after v14.

Attached my board  config. im only able to get things to work using the unofficial installer and the 4mb qspi verion on v14.

The tinys3 has 8mb on flash embeded on the esp chip and 8mb of external psram and it has 4 pins attached to the esp so I presume its in qio mode. Ill play around some more and maybe try qout. I guess technically the PSRAM is overkill and maybe even unused on WLED?  

### To Reproduce Bug

[env:um_tinys3]
;; ESP32-S3 development board, with 8MB FLASH and >= 8MB PSRAM (memory_type: qio_opi)
board = um_tinys3 ;; generic dev board; the next line adds PSRAM support
board_build.arduino.memory_type = qio_qspi    ;; use with PSRAM: 8MB or 16MB
platform = ${esp32s3.platform}
platform_packages = ${esp32s3.platform_packages}
upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} ${esp32s3.build_flags} -D WLED_RELEASE_NAME=\"ESP32-S3_tinys3\"
  -D CONFIG_LITTLEFS_FOR_IDF_3_2 -D WLED_WATCHDOG_TIMEOUT=0
  ;-D ARDUINO_USB_CDC_ON_BOOT=0  ;; -D ARDUINO_USB_MODE=1 ;; for boards with serial-to-USB chip
  -D ARDUINO_USB_CDC_ON_BOOT=1 -D ARDUINO_USB_MODE=1      ;; for boards with USB-OTG connector only (USBCDC or "TinyUSB")
  -DBOARD_HAS_PSRAM
  ${esp32.AR_build_flags}
lib_deps = ${esp32s3.lib_deps}
  ${esp32.AR_lib_deps}
board_build.partitions = ${esp32.large_partitions}
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.f_cpu = 240000000L
monitor_filters = esp32_exception_decoder
board_build.psram_type = qspi
board_upload.flash_size = 8MB

### Expected Behavior

should run wled.

### Install Method

Self-Compiled

### What version of WLED?

latest main branch v16-dev

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-12-30:
> board_build.arduino.memory_type = qio_qspi

@jesse-schein I don't know your board, however if it does have 8MB of PSRAM, then it should normally be `qio_opi` instead of qio_qspi.

You could even comment out the memory type line, to check if it works for you without PSRAM enabled.

> board_build.psram_type = qspi

The above line is not necessary, and it might actually break the build.

softhack007, 2024-12-30:
> It wont connect to wifi or show an access point after upgrading or after v14.

What messages do you see on serial monitor?

NickDolfin, 2025-01-17:
Hi Jesse,

I have similar problems with my esp32-S3-mini. No access point. 
I'm trying to figure out why.

Your report is 3 weeks old, did you manage to solve it?

UnexpectedMaker, 2025-03-24:
> > board_build.arduino.memory_type = qio_qspi
> 
> [@jesse-schein](https://github.com/jesse-schein) I don't know your board, however if it does have 8MB of PSRAM, then it should normally be `qio_opi` instead of qio_qspi.
> 
> You could even comment out the memory type line, to check if it works for you without adjusting any PSRAM options.
> 
> > board_build.psram_type = qspi
> 
> The above line is not necessary, and it might actually break the build.
> 
> > board_build.flash_mode = qio
> 
> this is the default from your [board definition file](https://github.com/platformio/platform-espressif32/blob/develop/boards/um_tinys3.json) - normally it's better to not repeat or duplicate these flags. Some cheap clones however may need "dio" or even "dout".

I know this is super late - but no, the 8MB of PSRAM is not OPI, it's QSPI. It's external PSRAM. All of my ESP32-S3 boards with 8MB of PSRAM use external PSRAM.
https://esp32s3.com