wled/WLED #4444: Reversing segment with spacing will not clear skipped LEDs

blazoncek, 2024-12-30:
### What happened?

When a segment with spacing > 0 is reversed, skipped LEDs are not cleared.

### To Reproduce Bug

Select segment, select Solid effect, set spacing to 0 and apply color (any).
Change spacing to 1 (or anything > 0) and select second (different) color.
Reverse segment and select third (different) color.
Some of the LEDs will remain painted with second color.

### Expected Behavior

Skipped LEDs are cleared when segment is reversed or mirrored.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32, ESP32-S3, ESP32-S2, ESP32-C3

### Relevant log/trace output

```shell
N/A
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2025-03-13:
Does this need to be added the the bugfix release?

blazoncek, 2025-03-13:
Yes. I think that's included in multifix.

PR #4568 (2025-03-11): Multifix
Plethora of fixes and tunings.
- some additional UI tooltips
- speed improvement in `fade_out()`
- tweaks in `color_blend()`, `color_add()` and `color_fade()`
- nested playlists (single level nesting) (fixes #4164 #3867)
- manual advance playlists
- bugfix in cleaning segment when spacing changed (fixes #4444)
- UI bugfix for entering W channel in Hex entry
- LED settings bugfix (incorrect handling of index wrap)
- replace some magic values with constants
- palette wrapping modification in `color_from_palette()` (WARNING may require some effects to change how they handle `wrap` parame [... cut]