wled/WLED #4445: v15 and playlists when using a scrolling text preset where text begins with a number

Timeline8, 2024-12-30:
### What happened?

When creating a playlist using the scrolling text effect, if the preset text starts with a number the playlist skips those presets. The one exception to this is if the numbered text preset is the very first one in the playlist, then only the first instance will play while all subsequent additions of it will be skipped. Switch off the playlist editor to show the API reveals that the subsequent additions of the same scrolling text preset got entered as 0. 

For example the playlist API shows

`{"playlist":{"ps":[8,2,0,3,0,4,0],"dur":[100,100,100,100,100,100,100],"transition":[7,7,7,7,7,7,7],"repeat":0,"end":0,"r":false},"on":true}`

where the [8,2,0,3,0,4,0] part should be [8,2,1,3,1,4,1] and manually changing the 0s to 1s fixes the problem.

### To Reproduce Bug

1. Create a scrolling text preset. I believe it may only do this if the text starts with a number. For example "1 Bug Text" will show as 0 while "Bug Text" will not.
2. Add some other effects to your presets (I was using all matrix effects of Soap, Sun Radiation, Waving Cell, and Pacifica).
3. Create playlist in the following order: Pacifica, "1 Bug Text", "1 Bug Text", Soap, "Bug Text" (no number is that last one). 
4. Save then reopen and unselect "Show playlist editor"
5. Order shown will be [6,0,0,2,7] - yours will have different numbers of course but the scrolling text ones with leading numbers will be shown as "0"
6. Run playlist and note that only, in this case, Pacifica, Soap, and "Bug Text" play while the two "1 Bug Text" presets are skipped.

[image]
[image]

You can go back into the API command window and replace the 0s with whatever number your "1 Bug Text" preset is and save and now you should see all 5 presets play.

After writing all the above, playing around with it some more and determining that if you start with the numbered text one, it will play the first, an even easier method to reproduce this is make the playlist simply "1 Bug Text" as many times in a row as you want and the API will look like [6,0,0,0,0,0].

### Expected Behavior

All presets added in the playlist editor should be added as non-0 in the API.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.15.0, build 241210, Environment ESP32-S2 v4.4.4 (0)

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

Wemos Lolin ESP32-S2 Mini board. Talking to it on a Mac running Chrome (both latest versions) using the web interface (i.e. typed in board IP address to get to the WLED built in menu. 
I did not try this through the phone app.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Timeline8, 2025-01-02:
Sorry new to WLED but what default value are you referring? In the playlist when adding a preset? Or do you mean a default value when creating a preset first? 

Are you saying there is a bug but not as I described? Or are you saying the problem lies in the way I am creating either the presets or the playlist and it not a bug?

Thanks.

blazoncek, 2025-01-02:
> Are you saying there is a bug but not as I described?

Yes. I was providing more insight for other collaborators.
Fixed in my fork with nested playlists commit. 13c4e423abca68fc86c70d72a34835a55cbfb682

softhack007, 2025-01-03:
> Fixed in my fork with nested playlists commit. [13c4e42](https://github.com/Aircoookie/WLED/commit/13c4e423abca68fc86c70d72a34835a55cbfb682)

@blazoncek your commit is rather large, and not good for cherry-picking into upstream. Can you give us a hint on how a minimal fix could be created?

blazoncek, 2025-01-03:
@softhack007 don't you want to use nested playlists? The minimal change needed is in index.js.

softhack007, 2025-01-04:
> @softhack007 don't you want to use nested playlists? The minimal change needed is in index.js.

@blazoncek not sure, it seems there is some nested-playlist-hype on discord? Personally I have no idea what is the use case for nested playlists (nested by a single level, or deeply nested?).

Independent from other playlist stuff under development, I think it would be very useful to have a minimal fix for the bug described in this problem report - i.e. something that is no longer than 10 line, and can be cherry-pickd into the next bugfix release of 0.15.0.1.

blazoncek, 2025-01-04:
It is a single level nesting but useful nevertheless. It is already possible to create nested playlists manually by editing playlist preset. There is one caveat, top playlist will not continue but it will restart in such case. Just ask @dosipod 

If you want single line fix, this is it: `plJson[p].ps.splice(i+1,0,1);` L: ~1938