wled/WLED #4446: Enable transitions unexpected behivior

dosipod, 2024-12-31:
### What happened?

Enable transitions and you will see leds speed up when changing brightness or other things , testing with DDP 
[image]

### To Reproduce Bug

Enable transitions 

### Expected Behavior

No sudden speed up 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.16.0-dev (build 2412040)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-31:
That is by design, during transitions, FPS limits are disabled. Not that I think its a good design but there appears to be some reason behind it. Maybe it is outdated. 
@softhack007 is this behaviour still needed after you updated FPS limit handling?

softhack007, 2024-12-31:
> @softhack007 is this behaviour still needed after you updated FPS limit handling?

🤔 good point. Actually it's not necessary. I think the reason for this speedup was the "solid" effect - it normally runs at 3fps so colors were flickering during brightness or color change. It might be better to change the effect so it does `return FRAMTIME` during transitions.

blazoncek, 2025-01-02:
FYI any effect that is not returning FRAMETIME will exhibit speedup/slowdown. That is bu design of the particular effect and the fact that during transition (if you want it to look smooth) strip has to update frequently.

blazoncek, 2025-01-07:
ICU is another example.