wled/WLED #4447: CHANGELOG wrong location

dosipod, 2025-01-01:
### What happened?

Due to renaming the branches , the release page CHANGELOGs  will give 404  
example https://github.com/Aircoookie/WLED/blob/0_15/CHANGELOG.md

### To Reproduce Bug

Click on any CHANGELOG  on the release page 

### Expected Behavior

No 404 

### Install Method

Binary from WLED.me

### What version of WLED?

any

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2025-01-03:
Can you share the link to the page that contains the wrong link rather than where the link is pointing to please

dosipod, 2025-01-03:
In the  releases page https://github.com/Aircoookie/WLED/releases 
under 0.15.0-rc.1  , 0.15.0-b7 , 0.15.0-b6 etc