wled/WLED #4448: IPv6 Support

apalrd, 2025-01-02:
**Is your feature request related to a problem? Please describe.**
WLED should support IPv6 completely

**Describe the solution you'd like**
- Receive IPv6 RAs and autoconfigure addresses accordingly (LLA + GUA if available)
- Support IPv6 sockets in any place an IPv4 socket is used
- Broadcast IPv6 AAAA records over mDNS (#3861 requests this)

**Describe alternatives you've considered**
In general almost all IPv6 features required are already provided by arduino_esp32 v3, however, they are not all present in the older versions currently used by WLED. It looks like version updates have been attempted in #4045 and therefore this can't be implemented until the version update is completed. 

Minor updates may be required to WLED logic which deals directly with IPAddress items.

netmindz, 2025-01-03:
As ipv6 usage in domestic settings is very low and ipv6 only rather than dual stack almost unheard of and without a killer feature of ipv6, I don't see this being added anytime soon.

netmindz, 2026-08-25:
Simple answer is that when doing development for such a resource limited platform, everything that is added means there is less resources for other features. @RedVyper 

Yes you can make option features, but then for example anything related to networking now has to be tested twice, once with that on and once off. So unless there is a killer feature or delivers significant benefit to a high percentage of users then it's going to be a no

RedVyper, 2026-08-25:
> Simple answer is that when doing development for such a resource limited platform, everything that is added means there is less resources for other features. @RedVyper

You mean resources on the limited board or limited resources in terms of human power to keep maintained the code?
If the former: there could also be a way to allow a IPv4-only stack (as of today) and an IPv6-only stack or a dual stack (because I'm convinced that the new ESPs can support a dual stack no problem).
If the latter: you can leave the issue as it is and wait for someone to PR an implementation.

And again, the "killer feature" **it is** the fact that there is support for the current Internet Protocol and not the legacy.

P.S. Closing an issue this way is not nice, especially if other people working on the project are agreeing with you but are still saying
> Maybe ipv6 will come at a later time.

netmindz, 2026-08-27:
People are welcome to create a PR, but as has been explained by 2 of the 4 maintainers of this project it's unlikely to be accepted and  definitely won't be enabled for default. 

Down voting is not helpful. There is still not a single reason *WHY* to add IPv6, what people can't do with v4 only, just the geeky "my network pure v6 is not a valid reason

apalrd, 2026-08-27:
Do you ever plan on supporting the Matter home automation protocol? IPv6 is not optional for Matter. Similarly, if you ever want to support Thread networking (and it's nearly a drop in feature on ESP32 platforms which have the radio for it), Thread cannot carry IPv4 packets at all.

CodeMystic21, 2026-08-27:
> it's unlikely to be accepted

I could maybe see the point for 8266.
But ESP32 has plenty of space. lwip already handle all the IPv6 network details.

The like 60 lines the PR proposed to just use a library provided function to **display** and assign addressen. Don't see how that would possible increase maintainance effort in any way.

RedVyper, 2026-08-27:
> Down voting is not helpful.

It actually is. It explains to other people how much a comment is agreeable with. Don't feel personally attacked. You as a person are doing a fantastic job in supporting and maintaining an open source project. But some of your ideas, especially as you exposed here, are definitely not agreeable with. Hence the downvotes.

> Do you ever plan on supporting the Matter home automation protocol? IPv6 is not optional for Matter. Similarly, if you ever want to support Thread networking (and it's nearly a drop in feature on ESP32 platforms which have the radio for it), Thread cannot carry IPv4 packets at all.

I suppose that you just gave them the "killer feature" they were requesting, after all.

apalrd, 2026-08-28:
Okay one more killer feature for you guys: Link Local Addresses + mDNS

Two devices connect at layer 2 (ethernet / wifi), link comes up, IPv6 automatically assigns a link local address (fe80::) on the interface. mDNS can now advertise this address, and clients can connect to it, regardless of what other addressing is going on (v4/v6). 

Client can now visit wled.local on their web browser. Regardless of how misconfigured any other IP configuration is (such as if the WLED or laptop has a static IP set in the wrong subnet, or the DHCP server is nonexistant because you plugged in directly to the laptop when you usually have it on the network, or you are connected in AP mode and the laptop has a static IP set), wled.local will continue to work, since mdns can use the link-local address and web browsers support using link local addresses when they are resolved over mdns. 

This doesn't require any change to your listening sockets, unless you've hardcoded AF_INET instead of AF_UNSPEC somewhere, which it doesn't look like you have. The LLA just has to exist, all of Espressif's backend will use it properly.

softhack007, 2026-08-28:
https://github.com/wled/WLED/pull/5796#issuecomment-5455284901

This tells me that IPv6/dual-stack would not be an easy task - it requires a re-design and additional abstraction for the complete network layer. I can't see how/when we would be able to do that.

@coderabbitAI are there estimates (in relevant arduino-esp32 or esp-idf documentation) about how much flash size would be added for IPv6 API, including websocket, UDP, TCP and AsyncWebserver?

softhack007, 2026-08-28:
> There's currently a pull request to start with address assignment, so that's definitely not 'no progress'.

Which PR are you talking about?

softhack007, 2026-08-28:
guys, an open word. You seem to be delighted to add 👍 and 👎 hand signs - how many of you have been developing embedded software for the esp32? Besides making signs with hands, why not use the same to start a PR? We'll need to modify the network architecture, so I'd prefer to exclude _coding agents_ from this challenge 😜 because we want something that is robust, understandable and maintainable.

Edit: this statement is not related to my previous question to AI about expected size increase. It is a more generic statement about what needs to be achieved, and that I prefer human software developers - instead of AI **coding** agents - to do it.

Edit2: coderabbit is a **code reviewing** AI tool, not a coding agent.

CodeMystic21, 2026-08-28:
I think @RedVyper made one but then deleted it again because the aversion received here.

> prefer to exclude coding agents from this challenge

And then literally just used an AI yourself ...

> The ESP-IDF lwIP documentation states that disabling CONFIG_LWIP_IPV6 saves approximately:

> 39 KB of firmware size

Don't seem to find wled disabling CONFIG_LWIP_IPV6. So the claimed 39 KB increase your AI claimed is not actually valid, as it's already there ...

softhack007, 2026-08-28:
@CodeMystic21 there is a big difference between using an AI for research and information, and using an AI agent for creating source code. AI generated source code is often low quality, it misses corner cases, and its generally not maintainable.

CodeMystic21, 2026-08-28:
> [@CodeMystic21](https://github.com/CodeMystic21) there is a big difference between using an AI for research and information, and using an AI agent for creating source code. AI generated source code is often low quality, it misses corner cases, and its generally not maintainable.

Doesn't change that it's still wrong or at least very misleading in both cases here.

If the reason for not including IPv6 included "39KB" space safe, this does not hold.

softhack007, 2026-08-28:
I was not talking about 39 KB. My intended message was "we need a good software architect (bio brain) to properly re-design the network layer".

CodeMystic21, 2026-08-28:
> 🧩 Analysis chain
> 🤖 web_search executed:
> 
> `ESP-IDF IPv6 flash size increase lwIP IPv6 documentation CONFIG_LWIP_IPV6`
> 
> Length of output: 10075
> 
> 🤖 web_search executed:
> 
> `site:docs.espressif.com ESP-IDF IPv6 binary size flash RAM lwIP`
> 
> Length of output: 9446
> 
> 🤖 web_search executed:
> 
> `Arduino ESP32 IPv6 WiFi UDP TCP AsyncWebServer WebSocket support documentation`
> 
> Length of output: 12687
> 
> `@softhack007`, yes. ESP-IDF publishes a relevant baseline estimate.
> 
> The ESP-IDF lwIP documentation states that disabling `CONFIG_LWIP_IPV6` saves approximately:
> 
> * **39 KB of firmware size**
> * **2 KB of RAM** when the system is powered up
> * **7 KB of RAM** while the TCP/IP stack runs
> 
> Therefore, for an otherwise equivalent **IPv4-only** build, enabling IPv6 has an approximate baseline cost of **39 KB flash** and up to **7 KB active RAM**. This is for the lwIP IPv6 stack in ESP-IDF. It is not a complete estimate for WLED dual-stack support.
> 
> ESP-IDF also states that IPv4 and IPv6 operate in a dual-stack configuration. It limits IPv6 address configuration to SLAAC and stateless DHCPv6 functions.
> 
> I found no published flash-size estimate for the additional IPv6 work in:
> 
> * Arduino-ESP32 wrappers
> * `AsyncTCP`
> * `ESPAsyncWebServer`
> * WebSocket support
> * WLED-specific address storage, formatting, discovery, UDP multicast, and test code
> 
> The final WLED increase can exceed 39 KB if those changes reference more code or require larger address structures and buffers. Conversely, linker garbage collection can remove unused code paths.
> 
> The documented **39 KB** is enough to validate the ESP8266 concern. A 40 KB-scale increase is material for a firmware that is close to a 1 MB application limit. For ESP32, the flash impact may be manageable on many configurations, but it does not remove the redesign and maintenance cost for dual-stack discovery and networking.
> 
> References: ESP-IDF [lwIP IPv6 support and minimum RAM usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html) and [binary-size guidance](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html).
> 
> You are interacting with an AI system.

so this message was just created for fun and not considered at all?

PS: Don't get me wrong: I also have my issues with AI slop ...

softhack007, 2026-08-28:
> so this message was just created for fun and not considered at all?

I still don't understand what you want. If we agree that AI genrated code is sometimes not the best idea, we seem to be on the same page.

Edit: just for clarity: https://github.com/wled/WLED/issues/4448#issuecomment-5455351116 and https://github.com/wled/WLED/issues/4448#issuecomment-5455473935 are two different thoughts. In the first I was wondering if firmware size would increase (and asked AI to do some research), in the second I gave you my understanding about a possible way forward, and it is not a simple "rewrite some API calls" job. For the second topic, I don't need AI slop in a potential PR source code, especially if a new network abstraction layer (architecture) might be needed.

RedVyper, 2026-08-28:
> My intended message was "we need a good software architect (bio brain) to properly re-design the network layer".

This message is totally different in attitude from what I could read in other comments. I'd love to work with people with this mindset.

This is to express that this thread is not only to complain, it's to discuss and improve the project.

softhack007, 2026-08-28:
> This message is totally different in attitude from what I could read in other comments. I'd love to work with people with this mindset.
> 
> This is to express that this thread is not only to complain, it's to discuss and improve the project.

@RedVyper thank you, I fully agree 😃

apalrd, 2026-08-28:
I'm not sure why you guys think this is such a huge tear up. Sockets are sockets, none of your socket code should care what address family they. All of the actual packet construction is done by lwip / esp-idf already, and so is dns / mdns. 

Taking a quick look:
- AsyncTCP appears to be properly passing through socket family
- ESPAsyncWebServer is not forcing a specific family or listen address, so this should default to all (unless Arduino has a weird default compared to normal socket semantics)
- I would expect that the ~60-line PR from @RedVyper (which was retracted) would result in the web UI working correctly over IPv6 without any further changes
- A few protocols in WLED use IPv4 broadcast, which would need to use multicast in IPv6. They should probably be using multicast in IPv4 anyway (especially over wifi), but that's a whole other issue unrelated to basic support for IPv6, and IPv6 multicast support could easily be separated into a general 'broadcast -> multicast cleanup' issue.

softhack007, 2026-08-28:
@apalrd I think there are many implicit assumptions all over our current codebase, like
* 0.0.0.0 is the "undefined" address
* you can store an ip adress in a uint32
* ip adresses can be parsed as %d.%d.%d.%d
* 192.168.x.y is the local "class d" subnet, 10.x.y.z is a local subnet
* 255.255.255.255 means broadcast
* 239.0.0.x means multitasking

Finding and adjusting all these, plus creating a dual-stack network class, plus scrutinising all network API call is what makes me think it's going to be a tear-up. WLED development in the last 10 years was only looking at IPv4,  and we've cut corners in many places to save code size.

apalrd, 2026-08-28:
I tried to go through the code in Github search and find where IPAddress was used, and build a mental model of what would be impacted for a 'basic' approach (ignoring the streaming UDP protocols). I'm not sure how many of those assumptions change the socket interface, and how many are only relevant to setting up IPv4 on the interface (which still needs to be done). None of these assumptions are necessarily wrong - 8.8.8.8 is still a valid DNS server, for example. 

I think the root of the challenge is that all of these socket / address related questions are already solved with zero impact on the application if you use the posix sockets API, which is implemented by lwip and a valid approach in ESP-IDF, but not what is used in the Arduino framework. Presumably they are backwards compatible to Arduino's WizNet origins? You aren't using the 'usual' socket calls like socket(), bind(), ... which handle address families for you, conversion to/from string, dns lookup returning multiple results of arbitrary address families, ...

RedVyper, 2026-08-28:
@apalrd You are right, I removed the PR because I thought I needed more time to propose a fully reviewed codebase. I will do it and then I will PR it back, this time as WIP.
Anyway, my PR wasn't about making the web server to listen on the IPv6 socket. Not yet, at least. It was "just" to show that the LwIP stack that runs in the board actually could start a LLA without doing anything more than a bunch or LoC because it's already builtin in the network stack. At least for the `esp32_idf_V5` environment. It's at least something. And doesn't break anything.

@softhack007 I have a basic "read-only" beginning of the implementation. I will try to push it ASAP. It won't fix this issue but it's a first step towards it.
P.S. I think that you should reopen this issue. Closing it was a mistake. Especially if you're open to PRs that will make progress to fix it.

softhack007, 2026-08-28:
@RedVyper ok you got me to it ;-) I'll re-open this ticket.

@netmindz while the core team is not actively working on the IPv6 "dual stack", maybe keeping this ticket open (and visible)will help our community members in bringing together their ideas.

softhack007, 2026-08-28:
For any PR, please make sure you maintain compatibility with the "IPv4 only" code. All additions should be enclosed into something like `#ifdef WLED_ENABLE_IPV6` ... `#endif` to ensure we don't disrupt other users.

It also means that the main objective of these WIP PR(s) is to get feedback from maintainers and to allow early testing by interested users.

RedVyper, 2026-08-28:
I will try to do my best. In the meanwhile, I hope that this is just the beginning.

[image]

RedVyper, 2026-08-29:
In the meanwhile I wait for the review of the first PR, this is a preview of today's results. I don't have my test network handing out GUAs, but if they were advertised you'd get them as well.

[image]

RedVyper, 2026-08-30:
Today there's another improvement. I will still keep it private until the first PR is reviewed.

```
$ curl -g -sS "http://[$BOARD_IPV6_ULA]/json/info" | jq '{ip6, ip6http}'
{
  "ip6": {
    "link_local": [
      "fe80::26dcc3ff:fe3b:f8a4"
    ],
    "unique_local": [
      "fd11:2ab:753:100:26dcc3ff:fe3b:f8a4"
    ]
  },
  "ip6http": true
}
```

This is obviously just the board being reachable on IPv6, not dealing with changing settings, yet. It's still a good start anyway for a weekend of work, IMO.
The field `ip6http` means "is the web server listening on IPv6?