wled/WLED #4449: Homeassistant Colorpicker issue

bastian187, 2025-01-02:
### What happened?

Turning on/off, changing brightness and selecting effect works from Homeassistant. Color-picking doesn't work with the new Homeassistant OS.

### To Reproduce Bug

Latest Homeassistant and WLED Version; ESP8266 D1 mini MCU

### Expected Behavior

Change of WLED color according to selected Color.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-02:
does this work on older versions of wled i.e. 14.4?

bastian187, 2025-01-02:
with Version 14.4 it is working flawlessly

blazoncek, 2025-01-04:
Please post actual command HomeAssistant sends to WLED.

closed 2025-05-12 by github-actions[bot]