wled/WLED #4452: WLED 0.15 broke the display

yousaf465, 2025-01-03:
### What happened?

With 0.15 update my display isn't working 

### To Reproduce Bug

update to 0.15

### Expected Behavior

update to 0.15

### Install Method

Self-Compiled

### What version of WLED?

0.15

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2025-01-03:
@yousaf465 can you elaborate a bit on the details of your hardware?
- what kind of LEDs
- how many outputs
- usermods?

How does it look when your display is broken?

yousaf465, 2025-01-03:
[image]

softhack007, 2025-01-03:
How many outputs, and how many LEDs?

(a screenshot of your LEDs settings page might help)

yousaf465, 2025-01-04:
Screen shots added for reference. 
[image]

[image]

![Screenshot_20250104-063525.jpg](https://github.com/user-attachments/assets/3441c7df-a2d4-4fdb-a7[image]

-063423.jpg …]()

[image]

[image]

blazoncek, 2025-01-04:
That looks like a very old version of 4LD usermod. You should update that as well.

yousaf465, 2025-01-04:
> That looks like a very old version of 4LD usermod. You should update that as well.

Any code for new 4LD?
I will be compiling using discord bot

Juaneitor3000, 2025-01-31:
Hello in WLED 0.15  i think the problem is associated with U8G2 library version, i get Four line display working with this setup on platformio_override.ini  on a esp32 C3,  if you force an old version example; U8g2@~2.35.30 it will work, at least for me, because if you work with newest versions, it compiles but stuck at boot.

  -D USERMOD_FOUR_LINE_DISPLAY
  -D FLD_TYPE=SSD1306_64
  -D I2CSCLPIN=6
  -D I2CSDAPIN=5

#CHECK YOUR LIB VERSION!! YOU SHOULD ADD
lib_deps = 

  olikraus/U8g2@2.35.30        
  Wire

The problem comes when i try to eneble the rotary encoder mod,     it just stuck at boot.    I cant find a way to make it work.

i WIll apreciate any help. Thanks.

blazoncek, 2025-01-31:
For me it is working with:
```
lib_deps = ${esp32.lib_deps}
  olikraus/U8g2
  Wire
```

Rotary encoder as well.

yousaf465, 2025-02-01:
> For me it is working with:
> 
> ```
> lib_deps = ${esp32.lib_deps}
>   olikraus/U8g2
>   Wire
> ```
> 
> Rotary encoder as well.

it is working on bot build too?

blazoncek, 2025-02-01:
I don't use bot. Compile using PIO or use provided binaries from @srg74 or @wled-install

yousaf465, 2025-02-01:
> I don't use bot. Compile using PIO or use provided binaries from [@srg74](https://github.com/srg74) or [@wled-install](https://github.com/wled-install)

can you provide the link to @srg74 bins? I had the old link.

wled-compile, 2025-02-01:
@yousaf465 Did you try https://wled-compile.github.io/ ? Does it work (with usermod "I2C/SPI 4 Line Display Usermod ALTernative (New)" selected)?

srg74, 2025-02-01:
Confirmed working SSD1306 SPI display with WLED 0.16.0-alpha (build 2501310) and olikraus/U8g2 @ ^2.34.22. Also compiled with olikraus/U8g2 @ ^2.36.4 and works too.

Juaneitor3000, 2025-02-01:
OK thanks,  i have test my build on an standar ESP32  and it works perfectly, both MODS: FOUR LINES DISPLAY ALT and ROTARY ENCODER,  even the AUTOSAVE MOD works all together, the problem comes when y try to use both on the ESP32 C3.    FOUR LINES DISPLAY MOD works well, but when i enable the ROTARY ENCODER MOD ALT,  the software build correctly but fails at startup, reboot constantly.  If i use FOUR LINES DISPLAY ALT alone works,  but if i use it with ROTARY ENCODER and /OR AUTOSAVE do not.

There is a way to startup these MOD as disabled at first boot?

srg74, 2025-02-01:
Most likely you using boot strapping pins. Otherwise everything should be fine. Check if you have pull up resistors on your rotary encoder.

yousaf465, 2025-02-02:
> [@yousaf465](https://github.com/yousaf465) Did you try https://wled-compile.github.io/ ? Does it work (with usermod "I2C/SPI 4 Line Display Usermod ALTernative (New)" selected)?

I tried it on a testing esp32. I think the old version was working. didn't  try the new version. 

not working on this though. 

[image]

blazoncek, 2025-02-02:
Dear everyone, Four line display and Rotary encoder usermods *are not broken* and *are working* in 0.15 (or 0.16). I have both of those on my desk and have been tested thoroughly.
If you are having issues with compiling please use Discord or WLED forum to get help with compiling.

closed 2025-02-02 by blazoncek

yousaf465, 2025-02-02:
> Dear everyone, Four line display and Rotary encoder usermods _are not broken_ and _are working_ in 0.15 (or 0.16). I have both of those on my desk and have been tested thoroughly. If you are having issues with compiling please use Discord or WLED forum to get help with compiling.

can you post link to working bins?