wled/WLED #4453: v0.15.0: Missing audio sync recieve for ESP8266

a-jey, 2025-01-03:
### What happened?

Release note says:
Audioreactive usermod included by default in all compatible builds (including full audio sync recieve for ESP8266 - Frequency and Volume)

After updating from 0.14.4 I can't find any option to activate audio sync receive on my ESP8266.

[image]

[image]

### To Reproduce Bug

Updating an ESP8266 from 0.14.4 to 0.15.0

### Expected Behavior

Finding an option to activate and setup audio sync receive from another device.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2025-01-03:
You have to use the 160 build , example is  WLED_0.15.0_ESP8266_160.bin

a-jey, 2025-01-04:
> You have to use the 160 build , example is WLED_0.15.0_ESP8266_160.bin

Thank you.
That wasn't clear from the Release Notes.
It's maybe worth a small add there.0

I've read from several users in several discussions that they got the same problem.

softhack007, 2025-01-04:
> It's maybe worth a small add there.0

@a-jey I've just added this to our release page: "For 8266, the 160mhz firmware binaries (for example WLED_0.15.0_ESP8266_160.bin) include audio sync receiving."

dosipod, 2025-01-04:
@softhack007  Not related to this issue but to the info on the releases page  , my understanding is that you just edit  like below 
[image]

So only  a maintainer can do that and no way to PR it? if so then might be also look at fixing this 
https://github.com/Aircoookie/WLED/issues/4447

netmindz, 2025-01-06:
Is there a reason it's only on this variant @softhack007  we use in our regular 8266 for MM

dosipod, 2025-01-06:
It was mentioned that it **may** impact performance ,https://discord.com/channels/473448917040758787/779395228624617512/1297293939971198977  . It was not clear what issues but blaz uses a lot of esp8266 so could be a point to consider  .

As of now there are reports of connectivity issues with esp8266 even without audio sync which  could be false positive or due to HA , HW quality ,network setup ..etc as usual which not new  ,  I did not notice such issues when testing using ddp esp8266 D1 mini or esp02 but only  confirmed issues on esp01 ( we only beta test esp8266 with DDP and do not use them in production  so can not be 100% sure with local led output   ) .

From above it may be a good idea for now  to keep audio on the  160MHz  bin only as in all cases it is a compile time build  flag  and when you say "we use in our regular 8266 for MM" are you really sure there are no issues in MM because I do not think a lot of guys use MM on esp8266 to start with and few are testing in details if any .