wled/WLED #4454: does not save new segment name, when editing and checking

mirasu, 2025-01-03:
### What happened?

I edit the name of the segments and click to confirm the change.  It seems to change and when I move around or restart the esp32 or esp 8266, with wled14 or wled15 it returns to the default names.

### To Reproduce Bug

I edit the name of the segments and click to confirm the change.  It seems to change and when I move around or restart the esp32 or esp 8266, with wled14 or wled15 it returns to the default names.

### Expected Behavior

Being able to rename the segments to easily identify which lights they belong to and be able to export them to home assistant.

For example I have 2 ws2812b strips, one on each side of the bed. each on a pin of the esp8266, I would like to be able to call segment 0 = Luz_cama_mia
segment1 = Woman_bed_light. 
And that is like that but when saving a change or restarting it changes to segment0 and segment1.

I don't know if I'm doing it wrong,

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14 y WLED 0.15

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mirasu, 2025-01-03:
Delete

softhack007, 2025-01-04:
Assuming that "delete" means "case closed".

closed 2025-01-04 by softhack007