wled/WLED #4458: On/off presset turns on RGB

BrickSolid28, 2025-01-04:
### What happened?

I’m using all RBGW lights. I have one preset to slowly turn on all the lights to white and one preset to slowly turn off all the lights. When the lights turn on with the preset with only the white channel, the RGB pixels light up for a a short period of time and then it switches to pure white channel. When turning off the lights the white channel slowly dims down to a low value then the RGB channel lights up dimly before everything turns off. This only happens with a preset. The on/off button makes the lights dim up to on and down to off correctly. This is most noticeable with the white channel but also happens with solid colors. 

### To Reproduce Bug

Turn on white channel only, create a new preset, set fade up to 2000ms, turn off all lights, select the turn on preset you made. 

Turn off all lights, create a new preset, set fade up to 2000ms, turn on all lights, select the turn off preset you made. 

### Expected Behavior

Slow fade up to white. Slow fade down to black without the RGB LED’s turning on. 

### Install Method

Binary from WLED.me

### What version of WLED?

15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Dig octa used

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2025-01-06:
Please post on and off presets.

BrickSolid28, 2025-01-06:
I don't know how to interpret this but here are both presets.  Thanks!

**Turn off**
{"on":false,"bri":128,"transition":20,"mainseg":0,"seg":[{"id":0,"start":0,"stop":888,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

**Turn on**
{"on":true,"bri":128,"transition":20,"mainseg":0,"seg":[{"id":0,"start":0,"stop":888,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"n":"","col":[[0,0,0,214],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

blazoncek, 2025-01-07:
Change the color in off effect to black (`"col":[[0,0,0,0],...]`).

BrickSolid28, 2025-01-07:
That worked. Thanks! 
The only other thing I’ve noticed is that it’s very jittery as it’s ramping up/down but still at very low light levels. I’d love to stretch the fade on/off to be longer but the jittering gets more noticeable the longer the fade is. Would adjusting the gamma help with this? Maybe there’s another setting?

DedeHai, 2025-01-07:
> Would adjusting the gamma help with this? Maybe there’s another setting?

try to disable gamma correction completely and see if it changes anything. digital LEDs are "only" 8bit resolution, at low brightness, that means not a lot of brightness steps. also try manually stepping brightness at lowest levels (use browser) to see the resolution.

BrickSolid28, 2025-01-08:
I tried lots of different gamma settings but it didn’t help. I decided to keep gamma brightness off. I turned up the brightness set point so there are more brightness steps and that did help smooth out the transition.

closed 2025-05-15 by github-actions[bot]