wled/WLED #4465: animARTrix not showing in Usermods config despite being visible in "Info" tab

twintersx, 2025-01-08:
### What happened?

Mod is not visible in Usermods config despite being visible in Info tab
[image]
[image]

### To Reproduce Bug

Add usermod build_flag -D USERMOD_ANIMARTRIX
Uncomment library dependency ;https://github.com/netmindz/animartrix.git#18bf17389e57c69f11bc8d04ebe1d215422c7fb7

### Expected Behavior

usermod header and checkbox should be visible in Usermod config

### Install Method

Self-Compiled

### What version of WLED?

WLED v0.16.0-alpha

### Which microcontroller/board are you seeing the problem on?

ESP32, ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2025-01-08:
Confirmed also from our side on both esp32c3 and classic esp32

softhack007, 2025-01-08:
How did you add the animartix usermod? It's not part of the usermods included in the Aircoookie/WLED repo.

softhack007, 2025-01-08:
animartix currently relies on a small extension from the MoonModules fork, which allows any usermod to have an "enabled" option without requiring to explicitly implement addToConfig() and readFromConfig().

https://github.com/MoonModules/WLED/blob/79c26db4fab4327813b2362519a0f5104ac035d3/wled00/fcn_declare.h#L311-L318

actually it might be worth bringing this small enhancement to  upstream - it allows to simplify usermods that do not need settings other than "enable".

netmindz, 2025-01-08:
Do the effects show in the list? There aren't actually any settings so it should be enabled by default

dosipod, 2025-01-08:
Yes the effects will show in the list  but I thought there will be enable/disable button in the usermod  same as in MM but as softhack mentioned it is not there .

netmindz, 2025-01-08:
In the version in AC the enabled value does nothing at all, so having the option to enable/disable in the settings would make no sense

dosipod, 2025-01-08:
@netmindz  I think  the option to enable/disable in the settings would make sense if you want to remove the effects from the list as there are too many but sure if you think so .It is just the inconstant UI behavior between MM and AC may lead  to confusion. This is the only usermod I could see with this behavior but could be others we missed  , at least adding a comment in the readme might do .

netmindz, 2025-01-08:
As it's not included in any of the AC builds, if you don't want the effects, just don't add the usermod surely? 

MM has a better way to handle enable/disable for all usermods, but this is isn't present in AC so would need to bring that over really before looking to add optional effect adding to this usermod

softhack007, 2025-01-08:
re-opring as an enhancement.