wled/WLED #4469: Gamma brightness control auto-deslects

adbern, 2025-01-09:
### What happened?

When gamma brightness control is selected and "saved" it auto unselects itself after "save" and also unselects gamma color correction. 
[image]

### To Reproduce Bug

Select both gamma color and gamma brightness correction - change gamma value to 1.0, then save. Check if gamma correction is still checked. 

### Expected Behavior

Both gamma color and brightness should remain selected after save. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-09:
Tested on 0.15, saving works.
any more details on your setup and installation method? did you upgrade from an earlier verison or is it a fresh install?

adbern, 2025-01-11:
Fresh install but I used the QuinLED DigOcta ethernet build: https://github.com/intermittech/QuinLED-Firmware/releases

It looks like if you set gamma to 1.0 it unchecks itself, but 1.1 doesn't? I'll try the non-Quin build and see if I can re-produce.

adbern, 2025-01-11:
I also don't really notice a difference in performance when I change the gamma value...

softhack007, 2025-01-12:
Gamma = 1.0 basicially means "no gamma".

Also gamma correction only applies to color you select from the color picker, to custom palettes, and to global brightness (in case you set "apply gamma for brightness"l. 

The standard palettes all have a fixed gamma of ~ 2.5 that can't be changed.