wled/WLED #4471: Setting value for "Percent" effect changes preset to "unknown"

andreas-bulling, 2025-01-09:
### What happened?

I have created a preset "soc" that uses the effect "Percent" for part of a LED strip to show our car's state of charge. The value I set in Home Assistant using

```
action: select.select_option
metadata: {}
data:
  option: soc
target:
  entity_id: select.wled_kitchen2_preset

action: number.set_value
data:
  value: "{{ states('sensor.car_battery') }}"
target:
  entity_id: number.wled_kitchen2_intensity
```

I noticed that while the selected preset is properly shown in Home Assistant after the first command, the second command makes wled change the preset to "unknown".

I am not sure if this is a bug but it's definitely not nice, at least not for the "Percent" effect (and potentnially others) because that's exactly what this effect is supposed to do. 

### To Reproduce Bug

see above

### Expected Behavior

The preset stays the same, even if the value/data/intensity is changed.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0 build 2412100

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2025-01-09:
I think that what you describe is expected behaviour.
Changing effect sliders actually means that what you see is not exactly the same as the last loaded preset, because effect options are part of a preset. @blazoncek please correct me if I'm wrong.

🤔 I still see the point that we might need to rethink our design, as some users seem to have a different idea of what a "preset" might be...

blazoncek, 2025-01-09:
Exactly. Preset is a *known state* (which also includes global brightness).

Sa such if you change effect parameters (sliders or color/palette) you are no longer using that preset. As a workaround, do not use preset but effect directly.

closed 2025-01-09 by andreas-bulling

andreas-bulling, 2025-01-10:
Thanks for the clarification/confirmation. I think, as a user, what strikes me as inconsistent in this case is that - sure, a preset is a known/defined state - but if I select an effect whose purpose is exactly to be dynamic (like "Percent") then I would expect to be able to change its value (percent in this case) without "leaving" the preset.

DedeHai, 2025-01-10:
> but if I select an effect whose purpose is exactly to be dynamic (like "Percent") then I would expect to be able to change its value (percent in this case) without "leaving" the preset.

you are not leaving the FX, but you are leaving the preset. those are two different things you got mixed up.