wled/WLED #4478: preset cycling with wiz remote will not start at expected start from cycle

Erik-JanWind, 2025-01-10:
### What happened?

API does not start from selected starting point ([P1=100&P2=156&PL=~]) but continues to select old input+1 if selected via Wiz remote, but if via AP (expected behavior) it start from 100 if previous preset was lower than lower limit in API.

### To Reproduce Bug

Presets:
Preset 1= Playlist with presets 100-156
Preset 4= [P1=100&P2=156&PL=~]

Wizz remote:
Select preset 1
Select preset 4-->goes to 100, select preset 4-->goes to 101, select preset 1-->goes to 1, select preset 4 goes tot **102** (or whatever last presetnumber is one+1)

AP via computer:
Select preset 1
Select preset 4-->goes to 100, select preset 4-->goes to 101, select preset 1-->goes to 1, select preset 4 goes tot **100**

### Expected Behavior

That the behavior of selecting presest with Wiz remote would be the same as via AP. AP "resets" the loop to starting point if you start out of bound, with the remote it start from last input and does not resets to starting point.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2025-01-13:
What is AP?

Use JSON API instead of HTTP API in preset 4. It may work better.

Erik-JanWind, 2025-01-13:
AP, was my way of saying Acces Point :-), meaning controlled via laptop (in my case). I made a screen recording, after I entered the JSON API: {"ps":"100~156~"}, but same behavior.  When you selected a preset via laptop, eg 1, then preset 4 (the next preset API) it goes to 100 (starting point). But if you select preset 1 with the wizremote, and then preset 4 it goes (only the first time to preset 100), but if you select preset 1 again via remote and then again preset 4, it goes to 101.  In my usecase this is unwanted behavior and unexpted, because it works like a charm when not controlled via wizremote.

blazoncek, 2025-01-13:
GUI uses a different call than a button does.

Erik-JanWind, 2025-01-13:
Allright, I did something that workable for me, but i still don't understand it. In order to reset the presets to 100 with a remote button press, I added preset 2= {"ps":"100~-156~-"} No matter where I am with the preset cycle, when pressing remote button 2, it goes to preset 100. From there I can use my preset 3: {"ps":"100~156~-"} (previous) and preset 4: {"ps":"100~156~"}(next) to navigate again between 100-156.

Background:
I use it for 8 suits on batteries, if for some reason, 1 suit goes out of range or misses a command, and therefor is in another preset. I can "reset" all suits to 100 and go from there again. 

Curious is: that if 1 altered the preset 2= {"ps":"100~-156~-"} to {"ps":"120~-156~-"} it still resets to preset 100 and not to 120. But I don't have an understanding in the coding, but it gets me excited to learn again. Anyway thanks for the push and I am sorry if I used the wrong channel, but I really thought it was a kind of bug. I will close it.

closed 2025-01-13 by Erik-JanWind