wled/WLED #4479: I2S problem ESP32-S3

NickDolfin, 2025-01-11:
### What happened?

I get a problem whem uploading to an esp32-s3-mini-1-N8.
It hangs and nothing happens.
The output of the terminal is:

--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (9) esp_littlefs: mount failed,  (-84)
E (9) esp_littlefs: Failed to initialize LittleFS
E (1354) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
Ada

[image]

### To Reproduce Bug

Just compile with the shown above platformio.ini entry and upload it into a esp32-s3-mini-1-N8 module.

### Expected Behavior

Well, similar behaviour to the working moonmodule firmware. That alternate firmware runs nicely and shows the pins bindings etc and I can get into the wifi of the esp32.

I'm not sure why thgis happens in wled-015.0 Is the boardtype correct? I tried using 'esp32-s3-devkitm-1' but then I get an error: 
Resolving esp32s3-mini1 dependencies...
UnknownBoard: Unknown board ID 'esp32-s3-devkitm-1'

That board entry does exists in platformio, but it does not function.

### Install Method

Self-Compiled

### What version of WLED?

WLED-0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Anything else?

Yes, Apparantly the little filesystem is not initialized. After 2 time restarting, that error message is no longer seen. At least that happens with the moonmodule firmware. I expect that same behaviour this wled-0.15.0.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-11:
this is more of a support question...
>That board entry does exists in platformio, but it does not function.
it does not exist in the version used by WLED. Why don't you use the env provided in the platformio.ini? 
`[env:esp32s3dev_8MB_opi]`

netmindz, 2025-01-11:
Unless you are using WLED_DEBUG the AC builds don't actually show anything useful on the serial port

closed 2025-01-13 by DedeHai