wled/WLED #4491: Apply at Boot preset Tick can not be unchecked

warmbeer92, 2025-01-14:
### What happened?

When saving a preset with a specific ID and the apply at boot tick on the apply at boot tick can‘t be changed After creating the preset. 

### To Reproduce Bug

If you untick the apply at boot After creating the preset and Save the settings it First shows that the changes we’re saved sucessfully but they are not if you Check the new settings of the preset. Furthermore if you delete the preset with the specific ID and create a new with the Same ID the and uncheck the apply at boot tick, the apply at boot tick is set After creating the preset. 

### Expected Behavior

Apply at boot tick should save your changes. Furthermore After deleting a specific preset ID all „old“ setting should be deleted. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2025-01-21:
You can unset boot preset in LED settings.

warmbeer92, 2025-01-21:
Yes, I can untick it in preset settings. Then a dialog pop up „ saving preset ID“. 

But After that the preset is still having the apply at boot ticked on! Somehow you can‘t deactivate the setting anymore, once it is set.

netmindz, 2025-01-22:
Yeah this "handy" way to set which is used on boot had only considered half the requirement. You should also be able to remove as well from the same place