wled/WLED #4496: Documentation error on kno.wled.ge UDP Sync page

Christanoid, 2025-01-16:
According to the Documentation there is no DNRGBW
[image]

Looking at the code dnrgbw is in fact implemented, just not documented, as visible in udp.cpp
```c++
} else if (udpIn[0] == 4 && packetSize > 7) //dnrgb
{
  unsigned id = ((udpIn[3] << 0) & 0xFF) + ((udpIn[2] << 8) & 0xFF00);
  for (size_t i = 4; i < packetSize -2 && id < totalLen; i += 3, id++)
  {
    setRealtimePixel(id, udpIn[i], udpIn[i+1], udpIn[i+2], 0);
  }
} else if (udpIn[0] == 5 && packetSize > 8) //dnrgbw
{
  unsigned id = ((udpIn[3] << 0) & 0xFF) + ((udpIn[2] << 8) & 0xFF00);
  for (size_t i = 4; i < packetSize -2 && id < totalLen; i += 4, id++)
  {
    setRealtimePixel(id, udpIn[i], udpIn[i+1], udpIn[i+2], udpIn[i+3]);
  }
}
```

Sadly I don't know where / if the documentation source code is set and can't find it, so this fyi instead. Sorry.

dosipod, 2025-01-17:
https://github.com/Aircoookie/WLED-Docs/pulls

Christanoid, 2025-01-17:
Thanks, will adapt it there after testing if it acutally works.

closed 2025-01-17 by Christanoid