wled/WLED #4500: Compilation Error

hackintoshlover, 2025-01-18:
### What happened?

Hi,
Thanks for the amazing project.
I want to enable "Animated_Staircase" usermod so i followed all the necessary steps.
But everytime i try to compile it gives me an error saying cannot open source files.
Attached are the images for the same.

[image]
[image]

### To Reproduce Bug

Clone the latest wled repository and compile using PlatformIO IDE.

### Expected Behavior

Compile Normally and generate output.

### Install Method

Self-Compiled

### What version of WLED?

Latest

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

softhack007, 2025-01-18:
@hackintoshlover this is a well-known symptom when you did not rebuild UI files. Since 0.15 these files are not part of the repo any more, as they caused 90% of all merge conflicts. 

Please look here for more info:
* https://kno.wled.ge/advanced/compiling-wled/#preparations
* https://github.com/Aircoookie/WLED/issues/4472#issuecomment-2580916842
* https://github.com/Aircoookie/WLED/issues/4379#issuecomment-2537014077

TL;DR open a command panel in VSCode, enter `npm install; npm run build`. Then select the buildenv for your board and click on "compile and upload` button.

softhack007, 2025-01-18:
I think this item should be added to the FAQ (in our KB).

closed 2025-01-18 by softhack007

netmindz, 2025-01-22:
We should just stop the build if the npm run build fails, so users can more clearly see the root cause of the issue

freddy1301, 2025-02-25:
Hey I am also not a developer and have no clue about node and all this shit. But I stumpled across the solution by accident!

Run npm run build -- -f in the project root and then try clicking the arrow again in vscode.
Found it here: https://kno.wled.ge/advanced/custom-features/
> The html_*.h files will only be created or updated if changes have been made to the wled00/data folder. If you still want to recreate the files, you can use this command: 
> npm run build -- -f