wled/WLED #4501: WLED 0.15.0 doesn't allow GPIOs 32 and 14 on wroom ESP32 and adds flickering

thost96, 2025-01-18:
### What happened?

With WLED Software Version 0.15.0 LEDs connected to GPIO 32 and GPIO 14 start flickering. Also the LED preferenz config doesn't allow to save anymore, as GPIO 32 and GPIO 14 aren't allowed anymore. 

### To Reproduce Bug

Use WS2814 LEDs on GPIO 14 and 32 with WLED Software 0.15.0. 

### Expected Behavior

No flickering and no error message using GPIO 32 and 14 like in WLED Software 0.14.4 

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

softhack007, 2025-01-18:
@thost96 gpio 32, 14 and 15 are the default pins for audioreactive; if you don't use audioreactive please go into usermod settings, and set all I2S pins to "undefined" ( -1 ). After saving, you should be able to use these pins for other purposes.

LEDs flickering could be related to this known problem:
* https://github.com/Aircoookie/WLED/issues/4389

thost96, 2025-01-24:
@softhack007 thanks for mentioning the Pins 32, 14 and 15. I've removed /unset these in the usermod settings are you said. Now the LED preference configuration isn't complaining anymore but GPIO 32 usage, but it's still flickering and not working. So downgrading again to 0.14.4. 

Would be nice for the future, if new thins were added to some default gpio pins, that these would be mentioned in the release notes.

ladyada, 2025-02-03:
https://github.com/Aircoookie/WLED/pull/4450 allows the extra pins to be used, its not in a release yet

thost96, 2025-06-07:
No stale, still active issue

willmmiles, 2025-06-07:
Flickering/glitching problems are being tracked in #4389.

closed 2025-06-07 by willmmiles