wled/WLED #4502: Fix Ethernet builds with V4

netmindz, 2025-01-18:
### What happened?

No target 

### To Reproduce Bug

Add env

### Expected Behavior

Working ethernet 

### Install Method

Self-Compiled

### What version of WLED?

0.16.0.dev

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-01-18:
[ethfreeze.txt](https://github.com/user-attachments/files/18464846/ethfreeze.txt)

kilrah, 2025-01-27:
I tried uncommenting the DISABLE_ESPNOW, doesn't help

troyhacks, 2025-01-28:
Quick test shows v3 works, so that's good. 

V4 appears to get the Ethernet IP and then doesn't spark the webserver, and somehow (in my case at least) falls back to using WiFi. 

Tried with my standard platform from WLED-MM (instead of Tasmota) and same results:

```
; platform                =  https://github.com/tasmota/platform-espressif32/releases/download/2023.06.02/platform-espressif32.zip
platform                =   platformio/espressif32@^6.9.0
platform_packages       =   framework-arduinoespressif32 @ platformio/framework-arduinoespressif32@3.20014.231204
```

192.168.1.158 is the Ethernet IP but it's still only responding on WiFi:

```
Connected! IP address: 192.168.1.158
Init STA interfaces
mDNS started
Network event: 4
Network event: 7
```

[image]

MAC address in Info is of WiFi, I don't know if this is a hint or just a bug. 

So seems like it's doing most things... but not sparking up the webserver.

troyhacks, 2025-01-28:
@netmindz IDF v4.0.0 and up uses ARDUINO_EVENT_ instead of SYSTEM_EVENT_

This PR should fix it up, but might need some logic for non-ESP32 boards for build reasons:

https://github.com/Aircoookie/WLED/pull/4526

softhack007, 2026-02-07:
@netmindz i'm not 100% xsure.
We have merged the changes proposed by @troyhacks, and there were also some other V4 specific fixes for ethernet.

> Use this Tasmota build which is based on IDF 4.4.6:
> ✅ https://github.com/tasmota/platform-espressif32/releases/download/2023.10.04/platform-espressif32.zip

This recommendation is still pending - I can add the platform version as an option in platdormio.ini, and only use it on the ``_eth`` build.

netmindz, 2026-02-07:
> are we sure this is not a "fix some, break some" deal?

Quite possibly, that's the same for any update to the platform. We need working Ethernet support for the release, so it's either change the version just for the Ethernet builds and they possibly have random other bugs or we upgrade all to the latest stable release and have consistency and fix anything that breaks