wled/WLED #4505: LEDs no longer run smoothly

warp735, 2025-01-19:
### What happened?

I have installed WLED on the garage and since a few firmware versions the animation no longer runs smoothly. This is particularly noticeable with “scanner” and the transitions.

Is this a setting issue or how can it be fixed? I don't think it's a general problem, otherwise I'm sure others would have noticed it already. 

I use an esp8266

[image]

### To Reproduce Bug

effect mode scanner

### Expected Behavior

runs smoothly

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.16.0-alpha

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

warp735, 2025-01-19:
ps: now i test 14.0 and evrything is ok

DedeHai, 2025-01-19:
checked with ESP8266, 500 LEDs, its buttery smooth.

warp735, 2025-01-19:
and how is that supposed to help me now? 

With 14.0 it also runs smoothly for me. I already suspected a wrong setting in post 1 that may have been added.

DedeHai, 2025-01-19:
you could give more details on your exact setup and your settings, there are a million and one options.

warp735, 2025-01-19:
[image]

DedeHai, 2025-01-19:
works as well with the settings provided. maybe try a flash erase and make sure its not caused by an external source like home assistant or other wifi traffic. try if it works when disconnected from wifi.

warp735, 2025-01-21:
Thank you. I'll give that a try.

closed 2025-05-28 by github-actions[bot]