wled/WLED #4512: Flashing LED bug in Sunrise effect

AdmiralRaccoon, 2025-01-22:
### What happened?

I have 2x60 WS2815 on each side of my bed, doing a sunrise effect the morning. 

I´m noticing multiple annoying flashes during the effect (Duration 5 minutes), and it seems that only 1 LED is causing this (The 1st?), I have tried to capture it in this video:

https://github.com/user-attachments/assets/22a31a36-534c-4c18-aca6-c6c87657b715

Next step I would try is to make seperate controllers for each strip, however, this should work with 2 outputs on 1 controller anyway?

Thank you and have nice day
Philipp

### To Reproduce Bug

Just running the effect again

### Expected Behavior

Making a smooth sunrise without flashes

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

This LED Controller is a GLEDOPTO ESP32 
https://www.gledopto.eu/Gledopto-WLED-LED-Controller-for-Digital-Light-Strips

I can´t tell if the bug exists in 0.14.4, as I directly flashed 0.15.0

Powersupply is 12V/5A, the Strips a powered directly from the controller, this is the LED Setup:

[image]

If anything else is needed, I´ll be happy to provide it.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-02-05:
regular ESP32 or one of the new ones?

AdmiralRaccoon, 2025-02-05:
just an ordinary ESP32 according to the specs:

[image]

DedeHai, 2025-02-06:
may be related to I2S output, which is a known bug. Try this: add a third output with >300 LEDs (unused, this is to force RMT output instead of I2S) and see if it still happens.

AdmiralRaccoon, 2025-02-06:
Thanks, you mean like this:

[image]

but only entering the actual amount of LEDs here?

[image]

willmmiles, 2025-06-06:
Possible duplicate of #4389 .