wled/WLED #4515: [Feature Request][WebUI][Audioreactive] Add an indicator to the UI if audio packets are received

autoantwort, 2025-01-22:
**Is your feature request related to a problem? Please describe.**
I am using the audioreactive usermod and the "receive" mode. It is not clear to me if data are received or not? 

**Describe the solution you'd like**
Add a indicator if packets are currently received or not. 

**Additional context**
It seems not to work because my network maybe supports no multicast (?). 

Should it work if I send the audio packets directly to the ESP and do not use multicast?

dosipod, 2025-01-22:
I think adding something to the info screen for audio sync is a great idea @softhack007 I hope you would consider that as I do not try to mess with the audio usermod when you are still here but I will try to do that for my own builds if you do not see that to be  a good idea .

@autoantwort Meanwhile you can use the FFT_SAMPLING_LOG build flag to check as you see in the below pic  ( Not sure that would work with netdebug though as I did not try that yet but will do  ) . Using other flags   such as  SR_DEBUG or  MIC_LOGGER does not seem to work for audio sync and not sure of there are other build flags for audio sync  .

[image]

willmmiles, 2025-01-23:
Look in the info panel beside "Audio Source".   If packets are arriving, it will read "UDP sound sync - receiving".  If not, it will instead say "UDP sound sync - idle", or indicate that audio data is being used from a local microphone source such as "I2C digital".

dosipod, 2025-01-23:
I think that is good enough and in all cases peek can also be used to also check  , i could not replicate issues in the network but having the sender down the "UDP sound sync - idle" was displayed correctly .

netmindz, 2025-01-23:
Closing as the feature is there. Please open a new issue in WLED-Docs if there is a gap or further clarification needed in the docs

netmindz, 2025-01-23:
@autoantwort you don't mention what you are using as your sender, but if you are using the wledsr server, then there is forthcoming feature that tries to work around the issue of multicast not working for some folk.

It's also worth noting that MoonModules WLED does not suffer the same issue. 

You can also try the V4 nighty build from this repo

closed 2025-01-23 by netmindz