wled/WLED #4517: WLED and Ubiquiti U7 pro incompatibility/problem?

wila91, 2025-01-24:
### What happened?

Im runing WLED  on ESP32 and its been running good for guite some time. **Just now updated my old asus wifi to Ubiquiti U7 pro**. There has been reports that there is issues whit IoT devises so this problem that im having is moastlikely **something to do whit Ubiquiti U7 pro not just WLED**. 
So the bug it self is that when ESP32 WLED is connected to Ubiquiti U7 PRO WLED will not respond to commands at all and will say connection to light failed. it will be unresponssive for some seconds untill it will be normal for about 10s and then it will hapen again in a loop.
Also if WLED devise is connected to Ubiquiti U7 PRO all effects will stop and resume after the frize. This will loop also. [https://www.dropbox.com/scl/fi/mgrmrbx0qwylwuzjkh57p/U7_pro_20250124_163348.mp4?rlkey=e7003q2h7wi9qo67recwchr4b&st=hdhti522&dl=0](url) Link to video off the problem

**Any ideas what could be causing this?**

### To Reproduce Bug

Ubiquiti U7 PRO and ESP32 connected. Running a effect will make the problem easy to see but you will not be able to controll anything when when the device gets stuck

### Expected Behavior

effect should not stop and conrolling the light should be possible

### Install Method

Binary from WLED.me

### What version of WLED?

Nightly Release 20250124 and WLED 0.14.0 "Hoshi" have the same issue

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

wila91, 2025-01-24:
did some testing and its possible to ping the device but it's not possible to control

wila91, 2025-01-25:
The video shows how at about 8 seconds the lights freeze and after a while they continue again (at about 14 seconds) and then freeze again at 25 seconds... the same thing continues. Whit my old wifi ap everything is works whitout problems

blazoncek, 2025-01-25:
There are plenty of WiFi settings in UniFi controller regarding enabled features. Please unselect all incompatible features like fast roaming, BSSID transition, meshing, etc.

netmindz, 2025-01-25:
Please can you also try the _V4 bin from the nightly to see if that's any better/different please?

wila91, 2025-01-25:
> There are plenty of WiFi settings in UniFi controller regarding enabled features. Please unselect all incompatible features like fast roaming, BSSID transition, meshing, etc.

I have tried many different WiFi settings... roaming, BSSID transition, meshing are all off. There is "Enhanced IoT Connectivity" and i have tried whit that on and off. I have tried changing all possible settings that I thought might affect or help.

I have also tried their latest and few others EA  fimawires (latest 7.2.22) for u7pro

wila91, 2025-01-25:
> Please can you also try the _V4 bin from the nightly to see if that's any better/different please?

Tested _V4 bin from the nightly and its the same. No change

wila91, 2025-01-25:
10 seconds working 7 seconds stuck. 

The speed remains the same whether there is a lot of WiFi traffic or not. Even if I turn off all other frequencies, the same thing remains. Only when I turn off the connection from the U7 Pro does the problem disappears. If I connect the WLED to my old Access Point and then its ok for the U7 Pro to be on just cant be connetion between u7pro and WLED.

wila91, 2025-01-25:
Had my 3 WLEDS running chase effect at the same time and I could see them all. They were getting stuck out of sync but at the same rate. So one or two could be stuck and others were running.

wila91, 2025-01-25:
Didn't mean to close

blazoncek, 2025-01-25:
I remember having issues on UniFi APs (WiFi5) which were resolved by fixing 2.4GHz channels on each of them and trying to get them as little overlapping as possible.

wila91, 2025-01-25:
> I remember having issues on UniFi APs (WiFi5) which were resolved by fixing 2.4GHz channels on each of them and trying to get them as little overlapping as possible.

I only run one AP and the neighbors' WiFi can barely reach inside my house. I have Zigbee devices but they are ZigBee channel 11 (around wifi ch 1) and my 2.4GHz Wifi network is ch 13.

Dont know how to make it better

blazoncek, 2025-01-25:
> my 2.4GHz Wifi network is ch 13

Move away from channel 13. It is known to cause issues. Use 1, 6, or 11.

wila91, 2025-01-25:
changed to 11. And its still the same

the channel has changed several times. It has mostly been on automatic, although now that I have tested more I have also changed manually

wila91, 2025-01-25:
[image]

Connention should not be that bad...
And ping was good ealyer and didn't stop when leds get stuck

wila91, 2025-01-26:
[image]

When the LEDs get stuck the arvage FPS shows 0

netmindz, 2025-01-26:
> > Please can you also try the _V4 bin from the nightly to see if that's any better/different please?
> 
> Tested _V4 bin from the nightly and its the same. No change

Can you also try a full erase and install of that env from platformIO so you also get the right bootloader please?

wila91, 2025-01-26:
After I completely reinstalled all 3 started working normally. Shouldn't have trusted the OTA

OTA is just so easy to use whitout having to disassemble anything.

I don't understand how change of the access point made all the problems to surface.

Anyhow. Now the lights are working again!

Thanks!

closed 2025-01-26 by wila91

netmindz, 2025-01-26:
I'm not 100% clear of exactly what is in the bootloader, but it does seem to do much more than just load the app

blazoncek, 2025-01-26:
> I'm not 100% clear of exactly what is in the bootloader, but it does seem to do much more than just load the app

It also initialises the hardware.

intermittech, 2025-02-05:
Next time something you could try is taking out or changing the configured NTP server, I've seen reports of similar freezes before caused by an unavailable or flaky NTP server being configured.

wila91, 2025-02-06:
NTP server might have been a issue. I had one setup and my new IoT network did not have internet connection. 

Let's see if I have time and enthusiasm, I might test it by setting up an NTP server now and see that happens.

wila91, 2025-02-09:
> Next time something you could try is taking out or changing the configured NTP server, I've seen reports of similar freezes before caused by an unavailable or flaky NTP server being configured.

Turned on "get time from NTP server" and its freezing again. My IoT network does not have internet access so it can't connect

blazoncek, 2025-02-09:
> My IoT network does not have internet access so it can't connect

What's the purpose of getting time then? Do you have private NTP server? If not then disable NTP. You will not be able to use internal timers or scheduled presets.

wila91, 2025-02-09:
Now I turned on NTP just for the sake of experimenting. Previously, I had use for it on the old WiFi and that's why it was turned on. When I switched to the new WiFi, I didn't even think about that setting and its possible effect on the lights.

NTP is off and there is no intention to turn it on.

To avoid someone else potentially having a similar problem, would it make sense to have an error message stating that the NTP server cannot be contacted and this may cause a hang/freezing? (or figure out why it hangs/freezes and fix it?)