wled/WLED #4519: "Send notifications on button press or IR" does not work when using "JSON remote"

V-R-E, 2025-01-25:
### What happened?

### **What happened?**
When "Send notifications on button press or IR" is enabled and "JSON remote" is selected for the IR remote WLED does not send UDP sync notifications for IR events, notifications are only sent on button presses.

### To Reproduce Bug

Enable "Send notifications on button press or IR"
Enable "IR GPIO" with "JSON remote"
Upload custom "ir.json" remote file
Use IR remote to trigger IR event

### Expected Behavior

UDP sync notification should be sent on IR events when using JSON remote

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

Tested on ESP32/ WT32-ETH01 with ethernet connection
UDP sync notification are sent on IR events when using "24-key RGB" remote
Custom ir.json file used shown below

`"0xD242628C": {
    "label": "Playlist 3",
    "pos": "1x2",
    "cmd": "&PL=3"
  },
"0xCC19A35B": {
    "label": "Playlist 4",
    "pos": "1x2",
    "cmd": "&PL=3"
  }`

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

blazoncek, 2025-01-25:
Please use JSON API in the `cmd` key.

V-R-E, 2025-01-25:
Thank you!
Just confirmed using the JSON API commands sends UDP sync notifications on IR events.
I will submit a pull request with an added node to the Infrared documentation page.

blazoncek, 2025-01-25:
Actually I consider this a bug in implementation as button should trigger correct notification regardless of API used.

blazoncek, 2025-01-27:
I have a possible fix ready. @V-R-E are you willing to test? If so, which binary do you need?
PM me on Discord.

V-R-E, 2025-01-27:
Yes, just sent you a message.

PR #4528 (2025-02-09): Fix for "Send notifications on button press or IR" does not work when using "JSON remote" #4519
- added JSON handling for irApplyToAllSelected

Fixes #4519
closed 2025-02-09 by blazoncek