wled/WLED #4521: ESP8266: network problems

1rfsNet, 2025-01-26:
### What happened?

I tried to install WLED on a new ESP8266. This worked, but I have massive problems with the network connection, as you can see in the following ping. There is no way to access the web interface.
I tried it with versions 0.15.0, 0.14.4, 0.14.3, 0.14.2 and 0.14.1, and the error pattern is identical everywhere. **There is no problem with version 0.14.0.** Everything works normally there. I was able to reproduce the problem with five other ESP8266s. Its always the same.
Another router was tested with the same result. The WLED-AP keeps opening and closing. But I can't open the web interface.

### To Reproduce Bug

Installation via install.wled.me and manually flashed. Result is identical. 

### Expected Behavior

I would expect the network connection to work. Since the ESP works fine with version 0.14.0, I can rule out problems with the ESP and the network. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.1 and higher

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Reply from 192.168.178.180: bytes=32 time=161ms TTL=255
Request timed out.
Reply from 192.168.178.180: bytes=32 time=138ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Reply from 192.168.178.15: Destination host unreachable.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.178.180: bytes=32 time=166ms TTL=255
Request timed out.
Reply from 192.168.178.180: bytes=32 time=143ms TTL=255
Reply from 192.168.178.180: bytes=32 time=187ms TTL=255
Reply from 192.168.178.180: bytes=32 time=426ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.178.180: bytes=32 time=1252ms TTL=255
Reply from 192.168.178.180: bytes=32 time=7ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.178.180: bytes=32 time=49ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.178.180: bytes=32 time=195ms TTL=255
Request timed out.
Reply from 192.168.178.180: bytes=32 time=169ms TTL=255
Reply from 192.168.178.180: bytes=32 time=136ms TTL=255
Reply from 192.168.178.180: bytes=32 time=241ms TTL=255
Reply from 192.168.178.180: bytes=32 time=140ms TTL=255
Request timed out.
Reply from 192.168.178.180: bytes=32 time=292ms TTL=255
Request timed out.
Reply from 192.168.178.180: bytes=32 time=263ms TTL=255
Request timed out.
```

### Anything else?

I have a lot of existing WLED instances on ESP8266 with versions 0.14.4 and 0.15.0. It seems to me that the problem only occurs with new installations. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-26:
so if all works on many of your devices, what makes you think this is a bug?
more likely bad hardware (which there is a lot of out there)

netmindz, 2025-01-26:
I think you misread the report @DedeHai 

It's only with the later releases he sees issues

Can you try the _compat build please @1rfsNet

DedeHai, 2025-01-26:
I did ineed misread, I understoot some devices to work.
chances are high that its a hardware issue though. @1rfsNet did you try with an added large capacitor on the 3.3V rail?

blazoncek, 2025-01-27:
Possible duplicate of: #3764, #3690, #2932

DedeHai, 2025-01-27:
> The ESP is connected to on my computer with USB. That shouldn't be a problem?!

source does not matter, if there is not enough capacitance, wifi will not work right. and most boards do not have enough for the high demand WLED puts on the chip.

1rfsNet, 2025-01-27:
> > The ESP is connected to on my computer with USB. That shouldn't be a problem?!
> 
> source does not matter, if there is not enough capacitance, wifi will not work right. and most boards do not have enough for the high demand WLED puts on the chip.

But that doesn't make sense. Why do I have 10 ESP8266 with 0.15.0 and 0.14.4 that were set up a long time ago and work fine, and new ones don't work, even though the ESPs are from the same batch? Sure, one may be broken, but not 5 pieces. Especially if it worked before, I erase the ESP and then reinstall it, then it no longer works. I can't explain it to myself.

DedeHai, 2025-01-27:
install using the web installer?

1rfsNet, 2025-01-27:
> install using the web installer?

yes mostly but also tried one to flash manually.

DedeHai, 2025-01-27:
try the hardware fix and report back.

DedeHai, 2025-01-30:
@Barabba11 you can open a new issue, but describe in detail how to reproduce because I just tried that and it does not happen on my board.

1rfsNet, 2025-01-30:
> 0.15.0 on ESP8266 wake up from st-by after 2-3 minutes if SNTP client is active, if I uncheck it it keeps turned off as desidered. I hope you will fix it in next release

This is not the same issue. I waited hours and nothing changes. 

> [@Barabba11](https://github.com/Barabba11) you can open a new issue, but describe in detail how to reproduce because I just tried that and it does not happen on my board.

I don't know what else to write, after all, I didn't do much more than use the WebInstaller (as usual). And that with several different ESP8266s.

Barabba11, 2025-02-04:
I've made some tests, zthe LED turn on even if NTP is unselected, and it start witha  reboot (the boot preset is chosen). Activate NTP just reduces the time on the next turn on/reboot, it can be only one minute if selected and offset time is set high. In my opinion the but is on the offline part of the program, somehow a mistake happens and it reboots.
I also noticed that if I unselect the scedules (I've never touched/set them) on the next reboot they are selected again, why?
I would add, chosing the time zone between CEEST and EEST is complicated and has no sense in my opinion, I don't know what they are and I can't find my time zone which is GMT+3h, so I leave GMT and I add 10800seconds, but it' not that smart. Why don't you indicate here just offset hours instead of those labels?
Thank you

Benoit221, 2025-02-09:
Hello,

I have the same problem. And effectively on the same board if i "update" to 0.14.0 from 0.15.0. Ping and wen are working. 

Best regards,
B LORAND

Benoit221, 2025-02-09:
Another important thing. Version 0.15.0 was perfectly working in ap mode. But not after i connect it to my ap. After i update to 0.14.0, settings are always present and wifi was working on my AP. I don't try to update to 0.15.0 via ota for now.

Best regards,
B LORAND

Barabba11, 2025-02-10:
Hi mate, my issue was solved flashing again by https://install.wled.me/ I strongly recomend you do it before loosing time finding a solution for your issues.
I've also recomend staff to notify with evidence to all users to flash web, on the download page. Chances are many users are not aware about that

Benoit221, 2025-02-10:
It's what i used.B LORAND -------- Message d'origine --------De : Barabba11 ***@***.***> Date : 10/02/2025  09:25  (GMT+01:00) À : wled-dev/WLED ***@***.***> Cc : Benoit LORAND ***@***.***>, Comment ***@***.***> Objet : Re: [wled-dev/WLED] ESP8266: network problems (Issue #4521) 
Hi mate, my issue was solved flashing again by https://install.wled.me/ I strongly recomend you do it before loosing time finding a solution for your issues.
I've also recomend staff to notify with evidence to all users to flash web, on the download page. Chances are many users are not aware about that

[
{
"target": "https://github.com/wled-dev/WLED/issues/4521#issuecomment-2647250480",
"url": "https://github.com/wled-dev/WLED/issues/4521#issuecomment-2647250480",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

splitbrain, 2025-02-13:
For what it's worth, I am having the same issue on a NodeMCU ESP8266. Downgrading to 14.0 makes the device work flawlessly again.

cozybear-dev, 2025-02-22:
Not sure what the cause is, but I was having intense problems as well. I tried practically all versions - including the compat variants. The only version that works OK for me is 14.0 on my ESP8266 based device. I ordered an ESP32 variant, to see how that performs. I was quite surprised with how bad WLED was performing given how popular it is.

Barabba11, 2025-02-23:
You'd try the 15.1 nightly beta, there are some fixed for network problems

NguyenAnHuy, 2025-03-10:
I have several ESP8266 and ESP32 devices. I noticed that the ESP8266 version 15.0 frequently disconnects or fails to load the UI. This issue is especially apparent when I connect it to HomeKit—if I use an iPhone to open WLED, the UI becomes inaccessible. I believe this is a software issue, as it's unlikely that all my ESP8266 devices are faulty.

netmindz, 2025-03-13:
For anyone commenting on issues, please confirm if you have tried the 8266_compat build to see how it compares

b2oow, 2025-04-02:
If you work with ESP8266 try 0.16 [Nightly Release 20250402](https://github.com/wled/WLED/releases/tag/nightly)
At first glance it seems to work as well as version 0.14.0 in terms of networking.

1rfsNet, 2025-04-27:
> For anyone commenting on issues, please confirm if you have tried the 8266_compat build to see how it compares

Unfortunately, I haven't had time to look into this issue for a while, but now I do. The 8266_compat build actually seems to work reliably, but I will continue to monitor it. At least it behaves differently from the default build. The web interface has been accessible without any problems for the last half hour.

closed 2025-09-01 by github-actions[bot]

oblivioncth, 2025-09-02:
I know closed but:

- The issue definitely is related to the network stack, and something with the device trying to connect to the target AP, as even when I MAC blacklisted the device from my AP so that it couldn't connect and it hosted its own SSID, the device was still unresponsive after connecting. I had to disable the SSID for the device to become responsive
- 15.1 _compat build restored it to working order. Did not try the _160 build as my device does not have a built-in USB port so it's not trivial to reflash if that fails.