wled/WLED #4523: Leds not responding after update to upper than 0.14.4

Dexter0007, 2025-01-26:
### What happened?

Update to every version upper than 0.14.4 cause random static colors. 

I have:
- custom board with ESP32-S
- WS2811 12V connected to GPIO02 by buffer(3V3=>5V) 

I checked on the oscilloscope that signal on LED is constant PWM. If added 10k resistor between LED PIN and GND signal was changed to PWM blocks with about 200ms intervals.

On version 0.14.4 everything works good without any changes on hardware or configuration.

### To Reproduce Bug

Update to every version upper than 0.14.4

### Expected Behavior

Working leds

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0/0.16.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-26:
What levelshifter? And why pwm for adressable leds?

Dexter0007, 2025-02-01:
74AHCT1G125
It's square wave with stable width and frequency. It's totally different like in 0.14.4 version with same PIN and settings.

Dexter0007, 2025-06-02:
.