wled/WLED #4525: WLED fails to connect to network with subnet mask greater than /24

SpeedyGoneZales, 2025-01-28:
### What happened?

WLED 0.15.0 fails to connect to a network with a subnet mask greater than /24 (255.255.255.0), e.g. /22 (255.255.252.0).
This is the case both with a manually set address and via DHCP.

### To Reproduce Bug

On router, create a network, e.g. 172.20.0.0/22, assign a block of DHCP leases (or don't) and assign that network to the WiFi.
The router shows WLED associating with the WiFi, but even with a static address, it is unable to communicate.

### Expected Behavior

I'd expect WLED to be able to communicate in all IPv4 address spaces.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
I tried to get logs, but according to https://github.com/Aircoookie/WLED/issues/1261 and https://wled.discourse.group/t/wled-not-connecting-to-wifi-where-are-logs/11716/6 this is not supported.
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

netmindz, 2025-01-28:
Please can you also try the esp32_v4 bin from the nightly release?

SpeedyGoneZales, 2025-01-29:
No immediate luck, again it gets no address from DHCP and is not reachable when a static IP is assigned.
Is there a way to tell whether the WiFi is connected (as opposed to an address assigned)?
There may be an additional issue with the WPA3 passphrase containing spaces - need more time to test.
Btw. the WiFi scan froze in the nightly when I tried it.

blazoncek, 2025-01-29:
FYI WLED does not use any special WiFi or Ethernet handling. You may need to open an issue for underlying Arduino's WiFi class or ETHClass libraries (those use Espressif proprietary WiFi implementation on ESP devices).

gcdinsmore, 2025-01-30:
I've noticed something similar.  But I've got a /24 subnet mask.  If I restart the esp32, it will normally connect.

ESP32-S3-Nano
WLED from master branch a few days ago
Aged, but standard WiFi router

SpeedyGoneZales, 2025-02-02:
Ok, it seems there is a compatibility issue with router firmware Mikrotik RouterOS versions 7.17.0 and 7.17.1 (a case is open under SUP-178146). 
WLED is connecting successfully to the /22 network using a backup device with 7.16.2. 
Can also confirm that special characters in the passphrase are not the culprit.

closed 2025-02-02 by SpeedyGoneZales