wled/WLED #4527: Dual Home Assistant setups causes WLEDs devices to go down

Johboh, 2025-01-29:
### What happened?

I have four devices running WLED 0.15. Three of these are ESP32-S2, and one is a ESP8266.
I mainly control these using Home Assistant and the WLED integration.
I have had this setup for years and it has been working well. 

Recently, I have been elaborating with running a second Home Assistant instance on a separate server, this is to provide a backup/fallback/redundancy if the main Home Assistant goes down (or is down for maintenance). The second instance does not run any (main) automations, it just provide an UI with all the lights to control (Ikea, shelly etc), including WLED using the WLED integration (and also run motion sensor automations if the primary device is down, but no automations that controls the WLEDs).

But once I started with second setup, randomly and sometimes within hours or after 24h+, some of the WLEDs become unresponsive, and they are no longer accessible via web. They have to be power cycled to be restored, and after that they run fine until it happens again. This occur on both the ESP32-S2 devices as well as the ESP8266 device.

I know this setup is not the ordinary one. I do would like to know what is causing this and if there is anything that can be solved. If there is any logs or other information needed, let me know!

Home Assistant 2025.1.0
WLED 0.15.0

### To Reproduce Bug

On two different servers/devices/computers, add WLED devices using the WLED integration. Then wait. Notice that some devices will become unavailable eventually.

### Expected Behavior

Devices that keeps running in the same way when there is only one Home Assistant instance.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-29:
it may be a queue overload. can you test if the nightly build is any different?

Johboh, 2025-01-29:
Just updated all of them to latest nightly, have wait and see how it works. I will report back.

netmindz, 2025-01-30:
You might also be wise to turn off the polling of devices in HA, especially on the second server

Johboh, 2025-02-01:
@DedeHai Great news! All lights have been working without any issues after moving to nightly (as of Jan 29) while running two Home Assistant instances, and still having polling enabled. I will close this issue as fixed, but I'm also curios what commit/s that addressed this issue. Is it [this one](https://github.com/Aircoookie/WLED/pull/4516) perhaps?

(unrelated: I disabled polling for some Frontier Silicon "smart radios" and they work much better now, so thanks for the tip @netmindz)

closed 2025-02-01 by Johboh

DedeHai, 2025-02-01:
> Is it [this one](https://github.com/Aircoookie/WLED/pull/4516) perhaps?

correct.

netmindz, 2025-02-01:
I would take this as confirmation that your changes in  https://github.com/Aircoookie/WLED/pull/4516 are working as expected 🎉