wled/WLED #4531: Valid GPIO pins are missing from the drop-down selectors on 15.0

booherbg, 2025-01-31:
### What happened?

Hi! I successfully flashed WLED 15.0 to my [Sparkle Motion Mini](https://www.adafruit.com/product/6160) board (ESP32). I can turn on the Audio Reactive mod but am not able to configure the onboard microphone (uses gpio pins 9, 10, 23) because in the GPIO drop-down menu, pins 6-11 are missing (from both the audio reactive config panel and the LED configuration panel). Pin 23 does show up in the drop-down menu.

I dropped back to WLED 14.4 and found that all the pins are there, and the mic works great for audio-reactive functionality!

Is there a way to add those pins to the drop-down menu in 15.x or is this something that needs to be done at compile time? Thanks! 

### To Reproduce Bug

Install WLED 15.0 from the official web installer, and verify that not all GPIO pins show up in the selector drop-down interface on both LED configuration panel and audio reactive usermod microphone configuration panel.

### Expected Behavior

I expect to see all available GPIO pins, such is the case in WLED 14.4

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-01-31:
try the nightly build, there was an update for chips that do not use those pins for internal flash.

booherbg, 2025-02-01:
Thank you. I actually got it wrong (my bad!) - 16.0-alpha nightly build is what I installed, not 14.4 (14.4 has a boot loop problem on this board). 

The only bug is that up/down scrolling on mobile is tough because dragging down, instead of scrolling, tries to refresh the page.

booherbg, 2025-02-01:
Is 16.0-alpha stable enough to use?

DedeHai, 2025-02-01:
> The only bug is that up/down scrolling on mobile is tough because dragging down, instead of scrolling, tries to refresh the page.

That is a known bug and cant be fixed AFAIK. Its an android thing.

Alpha is not tested that much, but should be stable

netmindz, 2025-02-01:
It's possibly worth trying to identify the fix that had been applied to the development branch and apply the bugfix for to the 0_15_0 branch (ai need to see if I can rename that branch, it should really have be called 0_15_release)

DedeHai, 2025-02-05:
@netmindz that would be the PR from adafruit, I am not sure there were other changes made to pin restrictions in 0.16

closed 2025-07-23 by github-actions[bot]