wled/WLED #4532: Time-controlled Enabled/Disabled Preset Reset after Reboot

mkormendy, 2025-01-31:
### What happened?

Changing Enabled/Disabled checkbox for any of the Time-controlled presets does not persist the state after the device is rebooted - they all reset to default enabled.

### To Reproduce Bug

Uncheck any of the enable/disable checkboxes for the Time-controlled presets, save the settings .. they persist. BUT, reboot the device,... they unexpectedly all reset to enabled.

### Expected Behavior

The enabled/disabled settings for the Time-controlled presets should persist after reboot, not reset the enabled/disabled state to the defaults of all presets to "on"/selected.

### Install Method

Binary from WLED.me

### What version of WLED?

Only tested with WLED 0.14.4+

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell

```

### Anything else?

Settings set and saved:
[image]

Settings reset after reboot:
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

DedeHai, 2025-02-05:
preset 0 means disabled, so the checkmark is not saved. put any other preset and it is saved.

mkormendy, 2025-02-05:
That's not intuitive for what the feature suggests.

DedeHai, 2025-02-05:
a lot of things in WLED are not intuitive, you are invited to provide an update.

mkormendy, 2025-02-05:
Oh! Interesting, I didn't think it was that easy.

DedeHai, 2025-02-05:
anyone can contribute, just code it and do a PR.