wled/WLED #4535: V0.15.0+ Raises TypeError on ESP32-ETH01

juldeus, 2025-02-02:
### What happened?

Raises: TypeError: Cannot read properties of null (reading 'classList')

Tried it on two ESPs. Now running on v0.14.4 to not have the error.

### To Reproduce Bug

Installed Via Webinstaller.

### Expected Behavior

No Error

### Install Method

Binary from WLED.me

### What version of WLED?

v0.15.0 /v0.16.0 alpha

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ranvirgorai, 2025-02-09:
facing the same issue. It works fine when connected for the first time, but on the second attempt, it stops working and shows the error mentioned above

simondib, 2025-02-21:
I had this issue when the data wire was connected to the wrong pin on my WT32-ETH01 board. Also have to make sure the Data GPIO value is correct in LED & Hardware setup.

kilrah, 2025-02-23:
You probably are using a LED output pin that was robbed by the audioreactive usermod, go to usermods and set all pins for AR to Unused.

rorosaurus, 2025-02-27:
Weird, I encountered this as well. I confirmed I wasn't using a pin conflicting with audioreactive usermod. But after a couple reboots the problem went away...

011V32, 2025-05-07:
I have the same problem.

softhack007, 2026-10-12:
cherry-picked #5205 to 0.15.6-beta

closed 2026-10-12 by softhack007