wled/WLED #4538: Show effect slider label when clicking on the icon, phone

Andro-Marian, 2025-02-03:
**Is your feature request related to a problem? Please describe.**
I can't see what sliders do on the phone.
So, when clicking on the icon show show me what controls.

**Describe the solution you'd like**
When clicking on the Icon to show the slider label.

**Describe alternatives you've considered**
I don't think there is.
If there would be a better solution to try.

**Additional context**
I don't understand why it's shows Freeze on the first slider and Toggle labels on second.

[image]

DedeHai, 2025-02-04:
it shows the slider label when you tap and hold the slider. why add it to the icon?

DedeHai, 2025-02-07:
> Is not showing.

works fine on android so you need to provide more details.

Andro-Marian, 2025-02-07:
Open in Firefox

DedeHai, 2025-02-07:
seems to be an issue with firefox on mobile devices.